BehaviorTree/BehaviorTree.CPP #469: an error when install behaviortree

lyd00116, 2022-11-11:
Hi!I encountered an error when I install it in the source code 
When I run make in the build folder, it didn't make successfully
here's the error:
/home/lyd/work/work0/behavior_tree_ws/BehaviorTree.CPP/include/behaviortree_cpp/utils/magic_enum.hpp:652:17: error: static assertion failed: magic_enum unsupported compiler (https://github.com/Neargye/magic_enum#compiler-compatibility).
   static_assert(supported<R>::value, "magic_enum unsupported compiler (https://github.com/Neargye/magic_enum#compiler-compatibility).");
                 ^~~~~~~~~~~~
tests/CMakeFiles/behaviortree_cpp_test.dir/build.make:446: recipe for target 'tests/CMakeFiles/behaviortree_cpp_test.dir/script_parser_test.cpp.o' failed
make[2]: *** [tests/CMakeFiles/behaviortree_cpp_test.dir/script_parser_test.cpp.o] Error 1
CMakeFiles/Makefile2:369: recipe for target 'tests/CMakeFiles/behaviortree_cpp_test.dir/all' failed
make[1]: *** [tests/CMakeFiles/behaviortree_cpp_test.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

lyd00116, 2022-11-11:
I use ubuntu18.04 to install it
also use ROS1    and I have installed  through  
 sudo apt-get install ros-$ROS_DISTRO-behaviortree-cpp-v3
I just wonder why I can't compile this source code successfully 
Than you!

facontidavide, 2022-11-11:
Version 4 uses multiple dependencies that are missing in Ubuntu 18.04

You will need to stay with version 3.8.

lyd00116, 2022-11-11:
> Version 4 uses multiple dependencies that are missing in Ubuntu 18.04
> 
> You will need to stay with version 3.8.

I just installed v3.8
and met the same error.I don't konw why

lyd00116, 2022-11-11:
And when I installed Groot
It showed that I didn't install behaviortree_v3:
/home/lyd/work/work0/groot_ws/Groot/bt_editor/bt_editor_base.h:11:10: fatal error: behaviortree_cpp_v3/bt_factory.h: No such file or directory
 #include <behaviortree_cpp_v3/bt_factory.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

In fact,I have installed behaviortree through ros and I have build a project successufully
Can you give me some answer? Thank you

lyd00116, 2022-11-11:
> > Version 4 uses multiple dependencies that are missing in Ubuntu 18.04
> > You will need to stay with version 3.8.
> 
> I just installed v3.8 and met the same error.I don't konw why

I can't installed successfully through source code
but I can install it through ros package

facontidavide, 2022-11-19:
this was solved in Groot

closed 2022-11-19 by facontidavide