BehaviorTree/BehaviorTree.CPP #470: Humble - Ubuntu 20.04 - Compile error

roncapat, 2022-11-11:
When building with:
```
colcon build --symlink-install --packages-select nav2_behavior_tree behaviortree_cpp_v3
```
I get:
```
Starting >>> behaviortree_cpp_v3
Finished <<< behaviortree_cpp_v3 [16.7s]                       
Starting >>> nav2_behavior_tree
[Processing: nav2_behavior_tree]                         
--- stderr: nav2_behavior_tree                                     
CMake Error at /home/roxy/ros2_ws/build/behaviortree_cpp_v3/behaviortree_cpp_v3Config.cmake:27 (include):
  include could not find requested file:

    /home/roxy/ros2_ws/build/behaviortree_cpp_v3/behaviortree_cpp_v3Targets.cmake
Call Stack (most recent call first):
  CMakeLists.txt:14 (find_package)

---
Failed   <<< nav2_behavior_tree [1min 40s, exited with code 1]

```

So this package builds but then on `nav2_behaviour_tree` include, something goes wrong. Could you help?

I checked out v3.8 branch

facontidavide, 2022-11-11:
I will look into it. Uvbuntu 22.04 I guess, right?

roncapat, 2022-11-11:
20.04 :)

adamconkey, 2022-11-22:
I also have this issue, Ubuntu 20.04 with ROS2 Galactic.

facontidavide, 2022-11-22:
I will try to address this ASAP

facontidavide, 2022-11-22:
This is what I did:
- Download the Docker image of Humble 
- Inside this pristine image, I clone the **nav2** repository.
-  Used **rosdep** to install dependencies; this installed `ros-humble-behaviortree-cpp-v3` 
- Invoked `colcon build --packages-up-to nav2_behavior_tree`

And it works just fine.

I also tried to clone in the workspace the **BehaviorTree.CPP** repository, to use that instead of the Dbian package.
**Then I checked out the branch v3.8**

It also works.

Could it be that you are using the master branch, instead of v3.8?

I don't see any issue on my side.

closed 2022-11-22 by facontidavide

adamconkey, 2022-11-22:
Thanks for looking into it. In my case I was building from source, I checked out branch `v3.8` in a ros2 workspace and tried to `colcon build` it and got the error when it tried to build my own package that had BehaviorTree.CPP as a dependency. I've since just abandoned building from source and used the `apt` install instead.

afrixs, 2024-04-23:
The same issue has just occured to me, here is what I did to make version [3.8.5](https://github.com/BehaviorTree/BehaviorTree.CPP/tree/2fb11758ea4b1ab1f9501fe1032e7c28684fe3f6) building successfully in Ubuntu 22.04/ROS Iron along with a dependent package (I also had to change C++ version to 17, which is beyond the scope of this issue, but may help you):
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 95b2663..1f36fcb 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -5,8 +5,8 @@ project(behaviortree_cpp_v3)
 set(CMAKE_CONFIG_PATH ${CMAKE_MODULE_PATH}  "${CMAKE_CURRENT_LIST_DIR}/cmake")
 list(APPEND CMAKE_MODULE_PATH "${CMAKE_CONFIG_PATH}")

-#---- Enable C++14 ----
-set(CMAKE_CXX_STANDARD 14)
+#---- Enable C++17 ----
+set(CMAKE_CXX_STANDARD 17)
 set(CMAKE_CXX_STANDARD_REQUIRED ON)

 if(MSVC)
@@ -248,6 +248,17 @@ if(ament_cmake_FOUND)
     set( BEHAVIOR_TREE_INC_DESTINATION   include )
     set( BEHAVIOR_TREE_BIN_DESTINATION   bin )

+    target_include_directories(${BEHAVIOR_TREE_LIBRARY} PUBLIC
+      $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>
+      $<INSTALL_INTERFACE:include> )
+    install(
+      DIRECTORY include/
+      DESTINATION include
+    )
+    include_directories(
+      include
+    )
+
     ament_export_include_directories(include)
     ament_export_libraries(${BEHAVIOR_TREE_LIBRARY})
     ament_package()
@@ -295,6 +306,12 @@ if(catkin_FOUND)
     LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
     RUNTIME DESTINATION ${CATKIN_GLOBAL_BIN_DESTINATION}
     )
+elseif(ament_cmake_FOUND)
+    INSTALL(TARGETS ${BEHAVIOR_TREE_LIBRARY}
+      ARCHIVE DESTINATION ${BEHAVIOR_TREE_LIB_DESTINATION}
+      LIBRARY DESTINATION ${BEHAVIOR_TREE_LIB_DESTINATION}
+      RUNTIME DESTINATION ${BEHAVIOR_TREE_BIN_DESTINATION}
+    )
 else()
     INSTALL(TARGETS ${BEHAVIOR_TREE_LIBRARY}
         EXPORT ${PROJECT_NAME}Targets
```