BehaviorTree/BehaviorTree.CPP #474: CMakeLists issue when porting form V3 to V4

thehummingbird, 2022-11-22:
Hi,
I had BT example behavior which works for V3 [here](https://github.com/thehummingbird/robotics_demos/tree/main/behavior_trees/simple_pick_demo)

To port this to V4, I installed the latest Bt library from master and updated the CMakeLists file to the following (apart from updating the head file in code)

```
cmake_minimum_required(VERSION 3.10.2)

project(bt_simple_pick)

set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
find_package(behaviortree_cpp)

add_executable(${PROJECT_NAME} "bt_demo.cpp")
target_link_libraries(${PROJECT_NAME} BT::behaviortree_cpp)
```

But when I try to generate makefile with cmake, I get the following issues -

```
CMake Error at CMakeLists.txt:9 (add_executable):
  Target "bt_simple_pick" links to target "Boost::coroutine" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:9 (add_executable):
  Target "bt_simple_pick" links to target "ament_index_cpp::ament_index_cpp"
  but the target was not found.  Perhaps a find_package() call is missing for
  an IMPORTED target, or an ALIAS target is missing?
```
Can someone please help me figure out what's missing in my process?

facontidavide, 2022-11-22:
I will be happy to help.
The code in the repository is still related to V3. Can you create a branch for V4 that I can test?

facontidavide, 2022-11-22:
Also, I can't find a **package.xml** in your link

thehummingbird, 2022-11-22:
Thanks for your prompt response Davide!

[Here](https://github.com/thehummingbird/robotics_demos/tree/bt_v4_simple_bt/behavior_trees/simple_pick_demo) is the branch for v4

No, this repo is for basic Bt behavior with no colcon/catkin. 
I'm generating the makefile with cmake and then using the executable I get from make. So I never needed package.xml for v3 as well

facontidavide, 2022-11-22:
What surprise me is this line:

> Target "bt_simple_pick" links to target "ament_index_cpp::ament_index_cpp"

It makes me think that you are trying to link to a version of BT.CPP that was build with ROS2.
When you say "I installed the latest Bt library", do you mean `sumod make install`?

Are you sure that you are not trying to link to a version in **/opt/ros** ?

facontidavide, 2022-11-22:
Fixed! Thanks for reporting

closed 2022-11-22 by facontidavide