BehaviorTree/BehaviorTree.CPP #478: Broken catkin build and debian package for ROS noetic

rcp1-beg, 2022-11-24:
Hi,

the released version 3.8.0 for ROS 1 noetic (20.04) broke the catkin build by providing two cmake files with the same name:

```
/lib/cmake/behaviortree_cpp_v3/behaviortree_cpp_v3Config.cmake <-- Generated by cmake configure_package_config_file
/share/behaviortree_cpp_v3/cmake/behaviortree_cpp_v3Config.cmake <-- Generated by catkin
```

Catkin will find for dependent packages the first one and detect that behaviortree_cpp_v3 is not a proper catkin package and does not provide the standardized package variables. Thus, one needs to manually link BT::behaviortree_cpp_v3 with this release.

The problem was introduced with #379 as far as I can tell. One could wrap the manual `configure_package_config_file` in the CMakeLists.txt with `if(NOT catkin_FOUND)` as simplistic fix.

Thanks for looking into it!

facontidavide, 2022-11-24:
Just to be sure, what is your level of confidence on the solution that you mention will work and it will not brake the behavior for others?

facontidavide, 2022-11-27:
Should be fixed. I am bloom-ing soon

closed 2022-11-27 by facontidavide

rcp1-beg, 2022-12-05:
Thank you very much for the fix. Sorry that I did not answer your question, I was on vacation...