BehaviorTree/BehaviorTree.CPP #481: Request: Behavior tree migration of 4.0 in ROS2 Humble.

Naraharirahul, 2022-12-02:
Hi there,

I'm currently using behavior tree version 3.8.0. I have recently came  across the scripting which is available in behavior tree 4.0.1 the feature which I wanted to integrate in my system. But I also use some of the behavior tree plugins used by ROS2 humble for my setup where they still are using behavior tree v3. May I know when would this migration happen in ROS2 humble? Any timeline for this?

Thanks,
Rahul

facontidavide, 2022-12-02:
Hi,

it is not clear to me what you mean here: "But I also use some of the behavior tree plugins used by ROS2 humble "

Which plugins? Nothing stop you from compiling BT.CPP from master, if you need to.

SteveMacenski, 2022-12-02:
What some thing is released in buying a raise to a distribution, it cannot break ABI/compatibility. It would need to be released under a different name. 

In terms of navigation, I haven’t looked at the changes between versions three and four. I don’t believe any of the new states and scripting features are things we’d likely use in navigation directly, but I’m not sure if we need to change anything about our current set up to be compatible with version four.

Naraharirahul, 2022-12-05:
I see. I have gone through the migration changes from 3.x version to 4.x over [here](https://www.behaviortree.dev/migration) and I feel the C++ class structure needs to be updated.
The class names have been updated in C++ and even with the `USE_V3_COMPATIBLE_NAMES` set to `ON` (as mentioned in the above migration changes), I'm getting errors such as below on my plugin codes.
`error: ‘NodeConfiguration’ in namespace ‘BT’ does not name a type`

Let me know if I'm actually missing something here or it's just the way it works and I need to update the code.

facontidavide, 2022-12-05:
The problem is ABI compatibility, that you are not supposed to break.
I don't see nav2 migrating, mostly for that reason. This means that you can use bt4 in a different process, but not in the same process where Nav2+bt3 is running.

Naraharirahul, 2022-12-05:
This makes sense. Does this mean that there won't be a path of Nav2 stack developing on behaviortree v4 and so on the later versions?
 For now I will try to find a workaround, but probably in future with few more changes/additions to the plugins on later versions this should happen at some point right? Or are we moving in two different directions.

Naraharirahul, 2022-12-06:
I was not planning to close this issue. I wanted to know the plans for the future with nav2 migration with behavior tree 4.0, If not now then when?

SteveMacenski, 2022-12-06:
Yes

closed 2023-06-07 by facontidavide