BehaviorTree/BehaviorTree.CPP #482: Suggestion: Use SubTree in a compact form

5730289021-NN, 2022-12-06:
In my work, there's some simple task like Gripping object. Our implementation required to set 2 coil of PLC aka. a gripping coil and a releasing coil. Let me show you the example:

```
<!-- Just a simple implementation of a "Grip" SubTree -->
<root BTCPP_format="4">
    <BehaviorTree ID="Grip">
        <Sequence>
            <Script code="ungrip_coil:=5" />
            <Script code="grip_coil:=6" />
            <Switch2 variable="{close}" case_1="True" case_2="False">
                <Sequence>
                    <!-- Grip -->
                    <SetIOOutput coil="{grip_coil}" />
                    <ResetIOOutput coil="{ungrip_coil}" />
                </Sequence>

                <Sequence>
                    <!-- Ungrip -->
                    <ResetIOOutput coil="{grip_coil}" />
                    <SetIOOutput coil="{ungrip_coil}" />
                </Sequence>

                <Sequence>
                    <!-- Ungrip -->
                    <ResetIOOutput coil="{grip_coil}" />
                    <SetIOOutput coil="{ungrip_coil}" />
                </Sequence>
            </Switch2>

            <Delay delay_msec="100">
                <AlwaysSuccess />
            </Delay>

        </Sequence>
    </BehaviorTree>
</root>
```

If this tree is already loaded, I think we should be able to use the compact representation like
`<Grip close="True"/> `

But with the current implementation, it is limited to
`<SubTree ID='Grip' close="True"/>`

I personally think that it'll be much more elegant if the SubTree could be used in compact form 

AFAIK the compact representation does only support action [Compact vs Explicit](https://www.behaviortree.dev/docs/learn-the-basics/xml_format#compact-vs-explicit-representation).

facontidavide, 2022-12-06:
I think it might be a good idea. I will consider it, thanks

OPerepadia, 2023-11-18:
I also like this idea. However, I can see one issue with it. Currently, it's possible to have a node and a subtree with the same name, and if the syntax is the same, it will not be possible to distinguish them. One solution I can think of is adding some prefix to the name of the tree, but after all, it might not be a big issue.