BehaviorTree/BehaviorTree.CPP #484: [Bug]: Including subtrees with the "ros_pkg" attribute is broken for ROS1

Lnsrini, 2022-12-08:
- For ROS1 packages, the `ros_pkg` attribute of the `<include>` tag can no longer be used to specify a path to a Subtree.
- [**This commit**](https://github.com/BehaviorTree/BehaviorTree.CPP/commit/a55c1cee06097327ec15793461d8237e63b3377f) is probably the one which we believe introduced the problem 
- Problem was found when we made the upgrade from v3.6.1 -> v3.8.1

Lnsrini, 2022-12-08:
Just now realized there is already an open PR for the fix: https://github.com/BehaviorTree/BehaviorTree.CPP/pull/483

Lnsrini, 2023-01-11:
Good day @facontidavide ! Do you have time to take a look at the linked [commit](https://github.com/BehaviorTree/BehaviorTree.CPP/commit/a55c1cee06097327ec15793461d8237e63b3377f) and [open PR](https://github.com/BehaviorTree/BehaviorTree.CPP/pull/483) for the fix?
This currently breaks existing behavior for ROS1 packages.

closed 2023-02-03 by facontidavide

rcp1-beg, 2023-02-07:
@facontidavide Would this fix be enough to reason a new release for ROS 1 (3.8.3)? :)