BehaviorTree/BehaviorTree.CPP #485: I can't find BT::StatefulAsyncAction.

jwson97, 2022-12-13:
Hi, I am new to Behavior tree so I follow the tutorials (v4.0).

In Tutorials - Basics 04. Reactive behaviors. 

`class MoveBaseAction : public BT::StatefulAsyncAction`

I got the error in `BT::StatefulAsyncAction. `

I tried to find BT::StatefulAsyncAction in http://docs.ros.org/en/kinetic/api/behaviortree_cpp_v3/html/namespaceBT.html

But I can't find BT::StatefulAsyncAction too. 

Please let me know if I'm doing something wrong or if I'm misunderstanding something.

grammers, 2023-01-24:
I have the same issue, to me, it appears as if BT::StatefulAsyncAction only exists in the tutorials (Reactive and Asynchronous behaviors, and Asynchronous Actions).

facontidavide, 2023-02-03:
Documentation fixed!

closed 2023-02-03 by facontidavide