BehaviorTree/BehaviorTree.CPP #487: Issues with std::shared_ptr objects in nested subtrees

gracideasg, 2022-12-27:
Hi there,

I have a setup with nested subtrees that leads to my ROS node crashing when I try to dereference a shared_ptr object at different "nestedness" levels of the tree. The setup can be represented as follows:

Main Tree
-Subtree I
  -Subtree A
    -Subtree 1
       -Subtree a
    -Subtree 2

The problem occurs when I try to access a shared_ptr containing sensor data within Condition Nodes in Subtree 1 and Subtree a. The tick propagates down to Subtree a's Condition Node and is able to progress back up to Subtree 1 upon success. The problem arises when a Condition Node in Subtree 1 is ticked - the same shared_ptr that was dereferenced in Subtree a is dereferenced again, and the nav2 bt navigator node crashes. When I remove these Condition Nodes, the tree is traversed without issue, so I am confident the Condition Nodes are where this issue lies. I should note the sensor shared_ptr is initialized on the blackboard - is something happening to the blackboard when traversing different nested levels of subtrees?

Any advice is much appreciated.

facontidavide, 2022-12-27:
Please, provide code that allow me to reproduce the error and I will be happy to fix it

gracideasg, 2022-12-28:
I was able to debug by explicitly passing through the sensor parameter to the subtrees - for some reason, sharing the blackboard alone was not sufficient.

closed 2022-12-28 by gracideasg