BehaviorTree/BehaviorTree.CPP #488: SequenceNode should not return RUNNING when child node return success

qZhang88, 2022-12-28:
I found SequenceNode returning RUNNING status when tree is tickOnce and first child return success while second child not ticked.

I used the newest code, and found these in source code of sequence_node.cpp.
``` C++
    case NodeStatus::SUCCESS: {
        current_child_idx_++;
        // Return the execution flow if the child is async,
        // to make this interruptable.
        if (requiresWakeUp() && prev_status == NodeStatus::IDLE &&
            current_child_idx_ < children_count)
        {
          emitWakeUpSignal();
          return NodeStatus::RUNNING;
        }
      }
```

facontidavide, 2023-02-03:
This is a complicate issue that I discuss here: https://www.behaviortree.dev/migration#preemptable-nodes-and-treetickroot

The problem I see is that some people prefer one behavior and others prefer the other behavior.

The only solution would be to have two implementations, **Sequence** and **PreemptableSequence**.

But which one should be used by default?
The former is "less surprising" for some users, but impossible to **halt** correctly in reactive behaviors

qZhang88, 2023-02-03:
> This is a complicate issue that I discuss here: https://www.behaviortree.dev/migration#preemptable-nodes-and-treetickroot
>
> The problem I see is that some people prefer one behavior and others prefer [...]

I think we should obey the design philosophy of **Sequence** just as it means in a default way. Since we may add many code between two ticks to handle other things, it's hard to distinct **RUNNING** in side a sequence node or between two continuous nodes. So I'd prefer **Sequence** by default.

facontidavide, 2023-02-03:
I see your point, BUT, in a complex tree, how would you know who is running, anyway.
It is equally opaque.

Can you give me a concrete use case, example where this is not working for you as expected? 
This might help moving the conversation forward

qZhang88, 2023-02-04:
The problem actually come from the sleep function. I used ROS sleep between two ticks and add some loger to trace tree node status to find out if tree works as I wanted. I see you have used **WakeUpSingal** to **not block**. However this may cause uncleareance of code logic. As we want sleep certain time between two ticks, which may be confusing, when it does not sleep.

And it would also be opaque :smile: in your **ReactiveSequence** case, do we really want to restart the **AbortCondition** for every **SyncAction** be successed?

As you mentioned **PreemptableSequence** is a good solution. And a new column is need for this table https://behaviortree.github.io/BehaviorTree.CPP/SequenceNode/

| Type of ControlNode | Child returns FAILURE              | Child returns RUNNING            | Child returns SUCCESS |
| :--------------------------- | ----------------------------------------- | ----------------------------------------- | -------------------------------- |
| Sequence                  | Return **FAILURE**, Restart first child | Return **RUNNING**, Tick again          | Tick next child                |
| ReactiveSequence    | Return **FAILURE**, Restart first child | Return **RUNNING**, Restart first child | Tick next child       |
| SequenceStar            | Return **RUNNING,** Tick again          | Return **RUNNING**, Tick again          | Tick next child       |
| PreemptableSequence | Return **FAILURE**, Restart             | Return **RUNNING**, Tick again          | Return **RUNNING**    |

If **PreemptableSequence** is default, it would return **RUNNING** when child returns **SUCCESS**. That's not what I want. The worst case would a combination of **ReactiveSequence** with **PreemptableSequence**, that is **ReactivePreemptableSequence** :cry: .

Or maybe we could add a parameter to control returning **RUNNING** or **SUCCESS**.

facontidavide, 2023-02-05:
> However this may cause uncleareance of code logic. As we want sleep certain time between two ticks, which may be confusing, when it does not sleep.

I think it is the opposite :slightly_smiling_face: 
The sleep I implemented is specifically designed to reduce latency, because it can be interrupted.
This allow the tree to react faster.

>do we really want to restart the AbortCondition for every SyncAction returning SUCCESS or Sequence synch_sequence_a returning SUCCESS?

This is **exactly** what I am trying to achieve :upside_down_face: 

>If PreemptableSequence is default, it would return RUNNING when child returns SUCCESS. That's not what I want. 

Why not? I still don't understand in practice how this affect you negatively.
If you had even a ASYNCH child, you will have to loop ( something like `tickWhileRunning` anyway.

> worst case would a combination of ReactiveSequence with PreemptableSequence, that is ReactivePreemptableSequence

## I think this is the key of this conversation:

Are you aware that in the specific case you mention, `ReactiveSequence` will **stop** being reactive and becomes equivalent to a Sequence??

qZhang88, 2023-02-08:
> I still don't understand in practice how this affect you negatively.

I have used ROS sleep, that's what got me in trouble. I will change to Tree::sleep() firstly.

Do you think it would be a better solution if there is a switch or flag in `Sequence`. Then we can control returning RUNNING or  directly ticking next child.

facontidavide, 2023-02-08:
yes, I think adding this parameter is a good idea. I will keep you posted on progress

msmcconnell, 2023-02-16:
Seems like the parameter is being added, so the issue will be resolved soon, but I am curious how ReactiveSequence could have ever been cleanly nested without this change? (new to the library so please let me know what I'm missing)
If it works currently as described above, then any time one reactive sequence was below another wouldn't it automatically halt all upper level siblings even if the lower ReativeSequence was going to return SUCCESS in the same tick? That would make subtrees hard to include and reason about without knowing their internal usage of ReactiveSequence. 

And to me it seems that would entirely prevent the Reactive Back-chaining design approach. An approach which is discussed extensively in BT literature for creating reactive trees. See Section 3.5 of Behavior Trees in Robotics and AI: An Introduction https://arxiv.org/pdf/1709.00084.pdf for an example of what I mean.

facontidavide, 2023-02-17:
Hi @msmcconnell 

adding that parameter is not a solution, in my opinion, it is substituting a potential pitfall with another. choose you venom.

Before saying anything, have a look at this:

```XML
<ReactiveSequence>
    <AbortCondition/>
    <Sequence>
        <SyncActionA/>
        <SyncActionB/>
        <SyncActionC/>
    <Sequence>
<ReactiveSequence> 
```

### As a user, do you expect the `AbortCondition` to be checked once or 3 times?

Do you expect Sequence to be an atomic operation or an operation that can be interrupted?

In my opinion, Sequence returning RUNNING is perfectly fine. it is just saying "hey, I haven't finished yet to do what I am doing, please tick me again", as any asynchronous operation does.

I looked at the example 3.5 in the document, and it is not related in anyway to what we are discussing here, because those examples do NOT have a **ReactiveSequence** and there is no concept of halting there.

More specifically... 
[image]

Look at this. Since these Sequences and Fallback are **not reactive**, but regular ones, with **both** implementations of Sequence in 3.8 and 4.0, the Conditions are only ticked **once**.

In other words, in **this particular example** you would not notice the difference in behavior nor execution speed.
So, it is not related to what we are discussing.

facontidavide, 2023-02-17:
> I am curious how ReactiveSequence could have ever been cleanly nested without this change? (new to the library so please let me know what I'm missing)
If it works currently as described above, then any time one reactive sequence was below another wouldn't it automatically halt all upper level siblings even if the lower ReativeSequence was going to return SUCCESS in the same tick? That would make subtrees hard to include and reason about without knowing their internal usage of ReactiveSequence.

I am having an hard time following your reasoning, sorry. Can you share a concrete example of a tree that, in your opinion , does something unexpected or a case we can see a "subttree hard to include"  ?

msmcconnell, 2023-02-17:
@facontidavide Thanks for the detailed response. I would expect AbortCondition to be ticked once per tree tick. 

In most of the literature I've seen Sequence and Fallback actually behave like the Reactive variants in this library (see section 1.3). However, I think my functional concern was unfounded. I've been creating some toy examples and haven't been able to get the behavior I was expecting based on this thread, so it everything seems to be working the way I preferred.

facontidavide, 2023-02-20:
>  I would expect AbortCondition to be ticked once per tree tick.

Indeed! In version 3.9 the toy example requires a SINGLE click and in version 4+ it requires 3 ticks to complete.

but since the root has no knowledge of the number of ticks an arbitrary tree requires  (and MUST not have it), we will always do "keep ticking as long as status is RUNNING".

If you have any specific example that you want to bring to my attention, let me know

closed 2023-02-20 by facontidavide