BehaviorTree/BehaviorTree.CPP #489: Port type missmatch between blackborad and subtree port

farshbafdoustar, 2023-01-02:
I have an ActionNode (Action1) that has input port with std::vector<int> type:

```
<BehaviorTree ID="Main">
   <SubTree ID="Subtree_A" states="1;2"/>
  </BehaviorTree>

<BehaviorTree ID="Subtree_A">
    <Sequence>
        <Action1 states="{states}"/>
    </Sequence>
  </BehaviorTree>
```
When I want to pass sth to this port through a wrapper Subtree, I got the following error:

```The creation of the tree failed because the port [states] was initially created with type [std::string] and, later type [std::vector<int, std::allocator<int> >] was used somewhere else.```

it seems that the blackboard item defined in Main tree is created by default with std::string type. Is there anyway to define the type for the blackboard items in advance or any workaround to handle such a situation?

facontidavide, 2023-01-02:
Let me see if I can replicate this issue locally

facontidavide, 2023-01-02:
is this the master branch (version 4.0.1?)

facontidavide, 2023-01-02:
fix pushed to master. thanks for reporting

closed 2023-01-02 by facontidavide

farshbafdoustar, 2023-01-02:
Wow, Thanks Davide!