BehaviorTree/BehaviorTree.CPP #492: When I build the System, I encounter the following erro

robertokcanale, 2023-01-06:
Hello, I am trying to run a simple project while following the documentation.

robertokcanale, 2023-01-06:
I get tht following code after running 
`cmake ..
`inside the build folder

```
-- Configuring done
CMake Error at CMakeLists.txt:8 (add_executable):
  Target "BT_tutorials" links to target "Threads::Threads" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:8 (add_executable):
  Target "BT_tutorials" links to target "Threads::Threads" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

yendiDev, 2023-01-06:
> I get tht following code after running `cmake .. `inside the build folder
>
> ```
> -- Configuring done
> CMake Error at CMakeLists.txt:8 (add_executable):
>   Target "BT_tutorials" links to target [...]

Same error

facontidavide, 2023-02-03:
Is this **main** branch? I will have a look

facontidavide, 2023-02-03:
Can your confirm your build system? catkin/ colcon/ vanilla cmake ?

facontidavide, 2023-03-03:
hello?

yendiDev, 2023-03-03:
Hi,
I have not worked on this in a while. 
I will boot up the project and communicate over the weekend

mnumanuyar, 2023-03-07:
I have the same error. this might be relevant : https://stackoverflow.com/questions/60227678/cmake-throws-error-during-generation-when-building-openpose-library

mnumanuyar, 2023-03-07:
tobe presice i get this error not during cmake of BT but during cmake of a dummy project that uses BT. error is :  
```
-- Configuring done
CMake Error at /usr/local/lib/cmake/behaviortree_cpp/behaviortree_cppTargets.cmake:61 (set_target_properties):
  The link interface of target "BT::behaviortree_cpp" contains:

    Threads::Threads

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /usr/local/lib/cmake/behaviortree_cpp/behaviortree_cppConfig.cmake:27 (include)
  CMakeLists.txt:6 (find_package)

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

facontidavide, 2023-03-07:
Pushed a solution

closed 2023-03-07 by facontidavide