BehaviorTree/BehaviorTree.CPP #494: How can I return to the previous state of my tree?

RodrigoCodeBernardo, 2023-01-12:
I'm sorry if my doubt is simple; I'm just starting in the world of behaviour trees.

I give a set of tasks to my robot, and based on these tasks, I will generate a plan; if the plan has not been generated yet, I call the operator to create the plan "CallOperatorToGenerateTasks" so far, so good. The problem I need to be managing to understand is. Initially, I checked if the plan was feasible ("PlanFeasible"). If not, I call the function to generate a new plan ("TryReplan"). After this, I execute the plan ("ExecutePlan"), provided the conditions are met. Here comes my question: If I have to interrupt my plan and generate a new one, I can't go back in the tree to the replan function ("TryReplan"). How can I resolve this issue?

```
<root BTCPP_format="4" >
    <!-- ////////// -->
    <BehaviorTree ID="BehaviorTree">
        <Sequence>
            <Fallback>
                <isCreatedTasks/>
                <RetryUntilSuccessful num_attempts="3">
                    <CallOperatorToGenerateTasks/>
                </RetryUntilSuccessful>                
            </Fallback>
            <Sequence>
                <Fallback>
                    <PlanFeasible/>
                    <RetryUntilSuccessful num_attempts="3">
                    <TryReplan/>
                </RetryUntilSuccessful>             
                </Fallback>
            </Sequence> 
            <ReactiveFallback> 
                <PlanFeasible/>
                <ExecutePlan/> 
            </ReactiveFallback>                         
        </Sequence>
  </BehaviorTree>
</root>
```

Thank you very much in advance for your help

RodrigoCodeBernardo, 2023-01-13:
What I am looking for is something like the package naviagatio2 from ROS created:

https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/plugins/control/recovery_node.cpp

closed 2023-01-19 by RodrigoCodeBernardo

facontidavide, 2023-01-19:
Did you find a solution?

RodrigoCodeBernardo, 2023-01-19:
Hi @facontidavide 

Yes, I found it.

I created a node like RecoveryNode which is featured in the "navigation2"  [(nav2_behavior_tree)](https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/plugins/control/recovery_node.cpp) library for ROS2.

Here is an example of a simplified tree, where I use the created node "RecoveryNode". That is, if my plan that was initially dispatched fails or is no longer valid (Check the condition "PlanNotFeasible"), it calls ("TryReplan") for replanning. In the case below, it tries to replan up to 3 times ("number_of_retries"), only then it cancels the plan. Note: after the replanning if done successfully, the dispatch of the new plan is done.

```
<root>
    <!-- ////////// -->
    <BehaviorTree ID="BehaviorTree">
        <Sequence>
            <Fallback>
                <isCreatedTasks/>
                <RetryUntilSuccessful num_attempts="3">
                	<CallOperatorToGenerateTasks/>
            	</RetryUntilSuccessful>                
            </Fallback>
            <RecoveryNode number_of_retries="3">           
            	  <ReactiveSequence> 
            	  	<PlanNotFeasible/>
            	  	<DispatchPlan/>             	  
            	  </ReactiveSequence>          	              	 
		  <TryReplan/>
            </RecoveryNode>                     	
        </Sequence>
  </BehaviorTree>
</root>
```