BehaviorTree/BehaviorTree.CPP #498: Ports with generic type `tf2::Stamped<tf2::Transform>` can't find the template specialization

torydebra, 2023-01-26:
Hi, I am following the [tutorial](https://www.behaviortree.dev/docs/tutorial-basics/tutorial_03_generic_ports/)  to parse a [`tf2::Stamped<tf2::Transform>`](https://docs.ros2.org/foxy/api/tf2/classtf2_1_1Stamped.html), which is included in the tf2 ROS library, but I got the error:
```
You (maybe indirectly) called BT::convertFromString() for type [tf2::Stamped<tf2::Transform>], but I can't find the template specialization.
```
and the throw: 
```
terminate called after throwing an instance of 'BT::RuntimeError'
  what():  missing required input [ref_T_goal]: You didn't implement the template specialization of convertFromString for this type: tf2::Stamped<tf2::Transform>
```

I have implemented the convertFromString in this way:
```c++
template <> inline tf2::Stamped<tf2::Transform> convertFromString(StringView str)
    { //never called because it is not found 
    }
```

The weird things are that if the port type is `tf2::Transform` it works:
```c++
template <> inline tf2::Transform convertFromString(StringView str)
    { //do stuff 
    }
```

and also it works if I remove the `inline` keyword:
```c++
template <> tf2::Stamped<tf2::Transform> convertFromString(StringView str)
    { //do stuff 
    }
```

Any clues about these behaviours?

---
fyi @fafux

facontidavide, 2023-02-03:
I think this is related to some problems with `inline` the way the linker finds the definition of the template specialization.

I believe it is a general problem with C++, not necessarily related to BT.CPP.

You may share with me a simple example, that I will be happy to review. Also, I think that there should be nothing special about `tf2::Stamped<tf2::Transform`, therefore you might be able to reproduce the issue with any `Foo` type (please confirm)

torydebra, 2023-02-06:
In my project, I am **not** able to reproduce the issue with any `Foo` type. I have a simple struct:
```c++
struct IndicesContainer {
    bool indices[6];
    IndicesContainer() : indices{true, true, true, true, true, true} {}
};
```
which gives me no problem at all with the `inline` option.

---

I have tried to reproduce the issue with the `tf2::Transform` starting from the tutorial, but this time I have no problem with inline, I attach the code anyway, (catkin workspace is necessary) 
[btGenericPorts.zip](https://github.com/BehaviorTree/BehaviorTree.CPP/files/10663572/btGenericPorts.zip)

I am not sure what I am doing differently in the code where I have the issue, I will double check it.

facontidavide, 2023-02-06:
I will have a look, but I am closing the issue because this looks like a regular C++ issue not related to the library

closed 2023-02-06 by facontidavide

galou, 2023-04-27:
For people who would stumble on the same issue (like me), I guess that the issue is due to the fact that the compiler exclude unused template specializations. I solved it by adding a dummy call to `convertFromString<name_your_type>()`.

galou, 2023-04-27:
In addition to my previous comment, it looks that explicit instantiation doesn't help.