BehaviorTree/BehaviorTree.CPP #499: Request: Documentation and/or Example of Switch Node

gracideasg, 2023-01-27:
I am interested in using the Switch node (Switch2, to be specific), but am unable to find any documentation on it apart from the header file switch_node.h. Could you provide a skeleton of what an xml file implementing this node would look like? I had no clue this node even existed until I saw it on Groot, since there is no mention of it in the nodes library on https://www.behaviortree.dev/.

facontidavide, 2023-02-03:
I am happy to help, but isn't this a sufficient explanation?

[StatefulActionNode](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/v4.0/include/behaviortree_cpp/controls/switch_node.h#L19-L41)

Of course, adding documentation on https://www.behaviortree.dev/ is good, but with a growing list of nodes, it is hard to cover everything.

facontidavide, 2023-02-03:
Additionally, since version 4.0, Switch node is kind of obsolete and easily replaceable with preconditions!
https://www.behaviortree.dev/docs/tutorial-advanced/pre_post_conditions

closed 2026-02-02 by facontidavide