BehaviorTree/BehaviorTree.CPP #504: Installation for Visual Studio

GeodeZister, 2023-02-09:
Hi everyone!! Sorry for my possibly stupid question, but please tell me how to properly install the BehaviorTree library in Visual Studio? Can't find the file BehaviorTree.h, I understand that you need to use the behaviortree_cpp folder?

mnumanuyar, 2023-03-07:
I am not sure about what you mean. Is this a problem about vscode intelisense and/or vscode cant find reference so that for example the line `#include "behaviortree_cpp/bt_factory.h"` has red squiglies under it? Or is it a more general question where you cant build your code?

facontidavide, 2023-03-14:
I think that this is not related to BT.CPP and miss any kind of information that may help solve an issue (if any).

Closing for the time being.