BehaviorTree/BehaviorTree.CPP #507: `_skipIf` precondition does not work with `SubTree`

alsora, 2023-02-16:
Hi, I'm using BehaviorTree.cpp version 4.0.1

I can successfully use the `_skipIf` precondition to avoid ticking some "standard nodes", but it seems to be ignored by the `SubTree`s. 
I suspect that the reason has to do with the fact that subtrees forward their arguments to the children nodes for remapping

See example
```
class DoSomething : public BT::SyncActionNode
{
public:
  DoSomething(const std::string& name, const BT::NodeConfig& config)
    : BT::SyncActionNode(name, config)
  { }

  static BT::PortsList providedPorts()
  {
    return {  };
  }

  BT::NodeStatus tick() override
  {
    std::cout << "Hello world "<< std::endl;
    return BT::NodeStatus::SUCCESS;
  }
};

int main()
{
    auto my_factory = std::make_unique<BT::BehaviorTreeFactory>();
    my_factory->registerNodeType<CheckFloat>("CheckFloat");

    std::stringstream bt_tree_text;
    bt_tree_text << R"(
    <root BTCPP_format="4">
    <BehaviorTree ID="MySubTree">
    <Sequence>
        <Action ID="DoSomething"/>
    </Sequence>
    </BehaviorTree>
    <BehaviorTree ID="MainTree">
    <Sequence>
        <Script code=" data:=true "/>
        <SubTree ID="MySubTree" _skipIf="data"/>
    </Sequence>
    </BehaviorTree>
    </root>)";

    my_factory->registerBehaviorTreeFromText(bt_tree_text.str());
    auto tree = my_factory->createTree("MainTree");

    tree.tickWhileRunning();
    
    return 0;
}
```

facontidavide, 2023-02-16:
thanks a lot @alsora , I shall look at this ASAP

facontidavide, 2023-02-17:
Fixed and unit test added. 
Wow, that was a big problem, thanks for reporting

[image]

closed 2023-02-17 by facontidavide

OPerepadia, 2023-02-17:
I was having this issue too, but thought it was something to do with my custom nodes. Thanks @alsora @facontidavide