BehaviorTree/BehaviorTree.CPP #511: Small improvements on the documentation

Gonzalo-Mier, 2023-02-27:
Hi, amazing project!!! I'm starting doing the tutorials and I found two small things that could be improved on the documentation.

- [On intro](https://www.behaviortree.dev/docs/intro), `The user can extend the "vocabulary" with his/her own custom nodes.` should be `The user can extend the "vocabulary" with **_their_** own custom nodes.` to be Non-Binary friendly.
- [On learn the basics -- decorators](https://www.behaviortree.dev/docs/learn-the-basics/BT_basics#decorators), the diagram has the node **`isDoorClosed`**, but it should be **`isDoorOpen`**.

I wish to have done a PR with this changes, but I couldn't find where the code is stored. Thanks for your time,
Gonzalo

facontidavide, 2023-03-08:
fixed. thanks

facontidavide, 2023-03-08:
For your information, code is stored here:  https://github.com/BehaviorTree/btcpp_website