BehaviorTree/BehaviorTree.CPP #515: "ReactiveSequence is not supposed to reach this point" happens when some children are skipped

torydebra, 2023-03-07:
Hi,
I am using a ReactiveSequence with `_skipIf` preconditions, and it throws:
```
terminate called after throwing an instance of 'BT::LogicError'
  what():  ReactiveSequence is not supposed to reach this point
```

Looking at the code https://github.com/BehaviorTree/BehaviorTree.CPP/blob/109eccd29f8cc4a872b27c8a64096589b3bd9f73/src/controls/reactive_sequence.cpp#L54-L72

it seems that this happens because if a children is skipped the `success_count` is not increased, am I right? 

I can provide simple example if necessary

facontidavide, 2023-03-08:
yes, a simple example would be really great. I will fix this as soon as I receive it

facontidavide, 2023-03-08:
Fixed. Thank for reporting

closed 2023-03-08 by facontidavide