BehaviorTree/BehaviorTree.CPP #523: Parallel Node: Is not a registered node

ephson24, 2023-03-13:
Hi all, this is how my Xml setup looks like:
`<BehaviorTree ID="GenericT">
    <Parallel failure_count="3" success_count="-1" ID="ParallelEx">
      <SubTree ID="UserActivityT"/>
      <SubTree ID="EmergencyT"/>
      <SubTree ID="PowerManagerT"/>
    </Parallel>
  </BehaviorTree>`

But every time I run the node, I get this error that says, "Parallel is not registered"
`Starting program: /home/novi/mobrob_ws/devel/lib/mobrob_agent/agent_node __name:=robot_behavior_node __log:=/home/ephson/.ros/log/c776fa04-c1a3-11ed-8920-652b62dfa80b/robot_behavior_node-1.log
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff3fa7700 (LWP 1505800)]
[New Thread 0x7ffff37a6700 (LWP 1505801)]
[New Thread 0x7ffff2fa5700 (LWP 1505802)]
[New Thread 0x7ffff27a4700 (LWP 1505803)]
terminate called after throwing an instance of 'BT::RuntimeError'
  what():  Parallel is not a registered node

Thread 1 "agent_node" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.`

facontidavide, 2023-03-13:
also, I am not aware that anything related to ENUMS doesn't work in master branch.
Can you also provide an example that fails related to ENUMS?

Thanks a lot!

ephson24, 2023-03-14:
> also, **I am not aware of any problem with ENUMS in master branch.** Can you also provide an example that fails related to ENUMS?
> 
> Thanks a lot!

In order to reproduce this issue:
**XML schema**
`<root BTCPP_format="4" >
  <BehaviorTree ID="PowerManagerT">
    <ReactiveSequence>
      <Script code=" deviceA:=BATT; deviceB:=CONTROLLER; LOW_BATT:=20 "/>
      <CheckLevel deviceType="{BATT}"
                  percentage="{LOW_BATT}"
                  isLowBattery="{isLowBattery}"/>
      <SendWarning deviceType="{BATT}" deviceStatus="{isLowBattery}"/>
    </ReactiveSequence>
  </BehaviorTree>
</root>` 

**CPP Code**
`#include <ros/ros.h>
#include <ros/package.h>
#include <behaviortree_cpp/behavior_tree.h>
#include <behaviortree_cpp/bt_factory.h>
#include <boost/bind.hpp>

enum DeviceType { BATT=1, CONTROLLER=2 };

BT::NodeStatus checkLevel(BT::TreeNode &self)
{
  double percent = self.getInput<double>("percentage").value();
  int devType = self.getInput<int>("deviceType").value();

  if(devType == DeviceType::BATT)
  {
    self.setOutput<bool>("isLowBattery", ((100 < percent)? true : false));
  }

  ROS_INFO_STREAM("Device| " << devType << " Level| " << percent);
  return BT::NodeStatus::SUCCESS;
}

BT::NodeStatus sendWarning(BT::TreeNode &self)
{
  int devType = self.getInput<int>("deviceType").value();
  bool devStatus = self.getInput<bool>("deviceStatus").value();
  //! Publish device warning on dedicated topic
  if(devType == DeviceType::BATT)
  {
    ROS_INFO("Processing warning update for battery ");
  }
  ROS_INFO_STREAM("Sending warning for | " << devType << " status: " << devStatus);
  return BT::NodeStatus::SUCCESS;
}

void registerNodes(BT::BehaviorTreeFactory &factory)
{
  factory.registerSimpleCondition("CheckLevel",
                               boost::bind(checkLevel, _1),
                               { BT::InputPort<double>("percentage"),
                                 BT::InputPort<int>("deviceType"),
                                 BT::OutputPort<bool>("isLowBattery")});
  factory.registerSimpleAction("SendWarning",
                               boost::bind(sendWarning, _1),
                               { BT::InputPort<int>("deviceType"),
                                 BT::InputPort<bool>("deviceStatus")});

  factory.registerScriptingEnums<DeviceType>(); //registering ENUMS here
}

int main(int argc, char** argv)
{
  ros::init(argc, argv, "robot_bt_cpp");
  ros::NodeHandle gnh, pnh("~");

  BT::BehaviorTreeFactory factory;
  registerNodes(factory);

  std::string bt_path = ros::package::getPath("mobrob_agent");
  auto tree = factory.createTreeFromFile(bt_path + "/config/test_bt.xml");

  BT::NodeStatus status = tree.tickOnce();
  while(not ros::isShuttingDown() /*&& (status == BT::NodeStatus::RUNNING)*/)
  {
    status = tree.tickOnce();
    tree.sleep(std::chrono::milliseconds(250));
  }
  return 0;
}
`

**Error Log**
`Starting program: /home/ephson/mobrob_ws/devel/lib/mobrob_agent/test_node __name:=robot_behavior_node __log:=/home/ephson/.ros/log/99f409c4-c233-11ed-8920-652b62dfa80b/robot_behavior_node-1.log
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff4090700 (LWP 1602351)]
[New Thread 0x7ffff388f700 (LWP 1602352)]
[New Thread 0x7ffff308e700 (LWP 1602353)]
[New Thread 0x7ffff288d700 (LWP 1602354)]
terminate called after throwing an instance of 'nonstd::expected_lite::bad_expected_access<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >'
  what():  bad_expected_access

Thread 1 "test_node" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.`

**+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++**
However modifying my xml file to look this (below) works flawlessly when using the literal 'LOW_BATT' without the ENUMS: BATT and CONTROLLER.
`<root BTCPP_format="4" >
  <BehaviorTree ID="PowerManagerT">
    <ReactiveSequence>
      <Script code=" deviceA:=BATT; deviceB:=CONTROLLER; LOW_BATT:=20 "/>
      <CheckLevel **deviceType="1"**
                  percentage="{LOW_BATT}"
                  isLowBattery="{isLowBattery}"/>
      <SendWarning **deviceType="1"** deviceStatus="{isLowBattery}"/>
    </ReactiveSequence>
  </BehaviorTree>
</root>`

ephson24, 2023-03-14:
> I am in the process of merging the changes of 4.1 into master. If you had a minimal example to reproduce the issue, I will include it to the unit tests

I figured out the Parallel Node issue. My main xml file was being overwritten by the export from Groot2 and hence the issue. It's all good now.

ephson24, 2023-03-14:
Totally unrelated issue (maybe deserves it own issue);  I find that **_skipIf doesn't work with ReactiveSequence** 
Using the same code as above and modifying the xml to look like this:
`<root BTCPP_format="4" >
  <BehaviorTree ID="PowerManagerT">
    <ReactiveSequence>
      <Script code=" deviceA:=BATT; deviceB:=CONTROLLER; LOW_BATT:=20 "/>
      <CheckLevel deviceType="1"
                  percentage="{LOW_BATT}"
                  isLowBattery="{isLowBattery}"/>
      **<SendWarning deviceType="1" deviceStatus="{isLowBattery}" _skipIf="!isLowBattery"/>**
      <SendWarning deviceType="1" deviceStatus="{isLowBattery}" />
    </ReactiveSequence>
  </BehaviorTree>
</root>
`

I get the **logic error**:
`terminate called after throwing an instance of 'BT::LogicError'
  what():  ReactiveSequence is not supposed to reach this point
Aborted (core dumped)
`

**Question**: Are pre-conditions/ post-conditions not meant to be used with certain control nodes?
**PS**: I didn't find anything that states otherwise neither in the documentation nor in the book by Petter Örgen and Michele Collendanchise.

facontidavide, 2023-03-14:
I am creating a unit test with your code related to ENUM, but it looks wrong.

b25480c296a6866c8ab302671031fb71ecf2cc27

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/tests/script_parser_test.cpp#L267-L296

I added an important fix (thanks!!!), but also there are errors in your code:

This is wrong:

```XML
<CheckLevel deviceType="{BATT}"  ... />
```
It should be wither one of these:
```XML
<CheckLevel deviceType="BATT" ... />
<CheckLevel deviceType="{deviceType}" ... />
```

Also, if you want informative error message, go not ignore the error state of `getInput`. You should write:

```c++
  DeviceType devType;
  if( auto res = self.getInput<DeviceType>("deviceType")) {
    devType = res.value();
  }
  else {
    throw std::runtime_error(res.error());
  }
```

facontidavide, 2023-03-14:
> I figured out the Parallel Node issue. My main xml file was being overwritten by the export from Groot2 and hence the issue. It's all good now.

Can you tell me more about that? If there is a problem in Groot2, I would like to fix it

facontidavide, 2023-03-14:
About `ReactiveSequence is not supposed to reach this point`, it should be fixed with a commit I did yesterday.

if you have any other issue, don't hesitate to tell me.

[image]

closed 2023-03-14 by facontidavide

ephson24, 2023-03-14:
> > I figured out the Parallel Node issue. My main xml file was being overwritten by the export from Groot2 and hence the issue. It's all good now.
> 
> Can you tell me more about that? If there is a problem in Groot2, I would like to fix it

Thanks a lot for the fixes. I am unable to reproduce this issue at the moment but I will keep an eye out for it and update this thread in the future should it come up.