BehaviorTree/BehaviorTree.CPP #524: Clarification Reactive Sequence, Conditions and Asynchronous Actions

gbcarlos, 2023-03-14:
I can't wrap my head around how Reactive Sequences have been implemented. 

I have the following simple example: 

[image]

#### What I want is:
- Battery Condition shall be checked after every Action that is executed
- The Battery Condition currently reads out the battery value from the Parameter Server and checks if it is higher than the limit
- All the actions return Success and are executed one after another as such
- For simplification, the Operator Actions are also implemented as Timers for now

In my example the condition is checked once, returns `True` and is not rechecked after every action. 

#### Question:

Is it possible to achieve this behaviour using a Condition? AFAIK Conditions only have the states `Success` & `Failure`, so the `ReactiveSequence` does not recheck because it has the returned `Success`?
Do I need to exchange the condition with an `Asynchronous Node`/`StatefulActionNode` and leave it in state `Running` until the battery drops under a certain level? Or should the condition be reset manually by one of the actions?

facontidavide, 2023-03-14:
> In my example the condition is checked once, returns True and is not rechecked after every action.

Let's start there, since this is a big problem. First, I am assuming that:

- you are using BT.CPP 3.8
- `timer` and `waitForOperation` are implemented as synchronous, blocking behavior and they never return **RUNNING**.

### Please confirm !!!!!!!!!

Note that:

- In the last 6 months all the development was focused on BT 4.x
- The very specific issue of a `ReactiveSequence` not working properly when ALL the children are synchronous has been debated a lot and fixed in version 4.
- This fix will not be ported back to version 3.8, because it may brake someone's code.

**Secondary:**

>Is it possible to achieve this behaviour using a Condition? AFAIK Conditions only have the states Success & Failure, so the ReactiveSequence does not recheck because it has the returned Success?

`batteryChargeBelow` **must** be a Condition and MUST NOT return RUNNING, to allow `ReactiveSequence` to work properly

> Do I need to exchange the condition with an Asynchronous Node/StatefulActionNode and leave it in state Running until the battery drops under a certain level? Or should the condition be reset manually by one of the actions?

You don't "need" to. That would be a different logic that would probably require a different tree, using a `Parallel` node instead of `ReactiveSequence`, since you would have a branch that wait for battery to be low before doing "something" and abort the other branch.

`ReactiveSequence` seems more straightforward.

gbcarlos, 2023-03-14:
Thanks for clarifying, yes, I am using v3.8.
Is the `Parallel` Node working as expected in v3.8? It might serve me as a solution for my overall problem.

facontidavide, 2023-03-14:
You don't answer the other important question. Are the `ReactiveNode` children synchronous or they may return RUNNING?

For reference: https://www.behaviortree.dev/migration#preemptable-nodes-and-treetickroot

> Is the Parallel Node working as expected in v3.8?
It SHOULD. 
No particular change was applied between version 3 and 4, so it either works in both or is broken in both.

If you see anything weird, let me know.

gbcarlos, 2023-03-14:
The children are all synchronous.

facontidavide, 2023-03-14:
ok, this is a wrong design. Something that is blocking must **NOT** be synchronous.

Having something that blocks and synchronous is a BUG.

`ReactiveSequence` makes sense only if you can interrupt the actions, and only asynchronous actions can be interrupted.

So, before moving to `Parallel`, I would fix that.  Cheers

closed 2023-03-14 by facontidavide

gbcarlos, 2023-03-14:
Good to know! Thanks for your support.