BehaviorTree/BehaviorTree.CPP #525: Pre-Condition : _skipIf NOT working with ReactiveSequence

ephson24, 2023-03-14:
Hi Davide,

I have this simple behavior to check battery level(condition always returning TRUE) and only send warning only if the "isLowBattery" variable is set. Thus, the **SendWarning** node should not run if it is false.
`<root BTCPP_format="4" >
  <BehaviorTree ID="PowerManagerT">
    <ReactiveSequence>
      <Script code=" LOW_BATT:=20 "/>
      <CheckLevel deviceType="BATT"
                  percentage="{LOW_BATT}"
                  isLowBattery="{isLowBattery}"/>
      <SendWarning deviceType="BATT" deviceStatus="{isLowBattery}" _skipIf="!isLowBattery"/>
    </ReactiveSequence>
  </BehaviorTree>
</root>`

But when running the tree, **SendWarning** keeps getting called regardless. However, if I change the control node to use a **Sequence**, it works as expected.

PS: I am using the same cpp code refrenced in [this issue](https://github.com/BehaviorTree/BehaviorTree.CPP/issues/523#issue-1621520744)
With the added code fixes when using **getInput**.

facontidavide, 2023-03-14:
Please look at this. it seems to work correctly to me:

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/tests/script_parser_test.cpp#L280-L288

Also, there was an error in your code that I corrected:

``` c++
 ((100 < percent)? true : false))
``` 
This will return false.
I changed it to: 

``` c++
 ( percent < 25 )
```

facontidavide, 2023-03-14:
Closing this. If you think that my unit test does not reproduce correctly your issue, please suggest a modification to the unit test and send a PR.

facontidavide, 2023-03-14:
I was able to reproduce the issue with much simpler code. Thanks a lot for your help!!!

https://github.com/BehaviorTree/BehaviorTree.CPP/commit/c71b2aaa759230d755d7d7ba1ec8056606e82565

facontidavide, 2023-03-14:
problem found, I think, it is a bug in **SimpleActionNode**.

I will keep you posted

facontidavide, 2023-03-14:
better unit test added. Have a look

closed 2023-03-14 by facontidavide

ephson24, 2023-03-20:
> better unit test added. Have a look

Will do. Thanks