BehaviorTree/BehaviorTree.CPP #532: ENUM| string conversion to int fails when int > 128

ephson24, 2023-03-21:
Hi Davide,

I don't know if this is the intended design but when using enums, the **conversion from string to int fails if the numbers used are greater than 128**.

In my code I have:
```c++
enum RobotNodeType { MAPPING=3000, NAVIGATION=3001, ODOMETRY=3002};
``` 
Which is being registered using:
```c++
factory.registerScriptingEnums<RobotNodeType>();
``` 

PS: I am currently being forced to use smaller values for the ENUM against my project design.

facontidavide, 2023-03-21:
Not intended. I will fix it ASAP

THANKS!

facontidavide, 2023-03-21:
Reason found: https://github.com/Neargye/magic_enum/blob/master/doc/limitations.md

The good news is that you can follow the instruction in this link to fix the issue.

Bad news is that I can't detect this error in **registerScriptingEnums**...  I will keep investigating

closed 2023-03-21 by facontidavide

ephson24, 2023-03-22:
Thanks for the link. Much appreciated!!