BehaviorTree/BehaviorTree.CPP #533: Sticky Pre-Condition flags

ephson24, 2023-03-21:
Hi Davide,

I use pre-conditions to skip executing certain nodes (node A) based on the required action. However the pre-Condition (post-condition down the line) flag can be modified by another node (node B) further down the line when it runs. Yes! node B returns either FAILURE or SUCCESS

The expected sequence on the next tick (given that the pre-condition/ post-condition flag has been set by node B) is that, node A should be executed. But for some reason, the pre-condition flag in node A is not updated and node A is skipped. Almost as if this is the first time the tree is being executed.

Is this behavior a bug or is that how pre-conditions are intended to function? 

**PS: I am ticking the tree in a while loop continuously**

facontidavide, 2023-03-21:
The expected behavior is that pre-conditions are checked only when transitioning from IDLE.

Can you provide a minimal failing unit test, or at least a simple XML that reproduce the error with simple Nodes?

facontidavide, 2023-03-21:
I created this unit test, but I can not reproduce your issue (it works as expected)

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/issue_533/tests/gtest_preconditions.cpp#L142-L177

facontidavide, 2023-03-27:
any update on this? Do you have a way for me to replicate the issue?

ephson24, 2023-03-28:
Hi Davide,
Here's the code to replicate this issue.

**CPP code**
```c++
#include <ros/ros.h>
#include <ros/package.h>
#include <behaviortree_cpp/behavior_tree.h>
#include <behaviortree_cpp/bt_factory.h>
#include <boost/bind.hpp>

enum DeviceType { BATT=1, CONTROLLER=2 };

BT::NodeStatus checkLevel(BT::TreeNode &self)
{
  double percent;
  if(auto res = self.getInput<double>("percentage"))
    percent = res.value();
  else
    throw std::runtime_error(res.error());

  DeviceType devType;
  if(auto res = self.getInput<DeviceType>("deviceType"))
    devType = res.value();
  else
    throw std::runtime_error(res.error());

  if(devType == DeviceType::BATT)
  {
    self.setOutput<bool>("isLowBattery", (50 < percent));
  }

  ROS_INFO_STREAM("Device| " << devType << " Level| " << percent);
  return BT::NodeStatus::SUCCESS;
}

BT::NodeStatus sendWarning(BT::TreeNode &self)
{
  DeviceType devType;
  if(auto res = self.getInput<DeviceType>("deviceType"))
    devType = res.value();
  else
    throw std::runtime_error(res.error());
  bool devStatus = self.getInput<bool>("deviceStatus").value();
  //! Publish device warning on dedicated topic
  if(devType == DeviceType::BATT)
  {
    ROS_INFO("Processing warning update for battery ");
  }
  ROS_INFO_STREAM("Sending warning for | " << devType << " status: " << devStatus);
  return BT::NodeStatus::SUCCESS;
}

void registerNodes(BT::BehaviorTreeFactory &factory)
{
  factory.registerSimpleCondition("CheckLevel",
                               boost::bind(checkLevel, _1),
                               { BT::InputPort<double>("percentage"),
                                 BT::InputPort<int>("deviceType"),
                                 BT::OutputPort<bool>("isLowBattery")});
  factory.registerSimpleAction("SendWarning",
                               boost::bind(sendWarning, _1),
                               { BT::InputPort<int>("deviceType"),
                                 BT::InputPort<bool>("deviceStatus")});

  factory.registerScriptingEnums<DeviceType>(); //registering ENUMS here
}

int main(int argc, char** argv)
{
  ros::init(argc, argv, "robot_bt_cpp");
  ros::NodeHandle gnh, pnh("~");

  BT::BehaviorTreeFactory factory;
  registerNodes(factory);

  std::string bt_path = ros::package::getPath("mobrob_agent");
  auto tree = factory.createTreeFromFile(bt_path + "/config/test_bt.xml");

  try {
    BT::NodeStatus status;
    while(not ros::isShuttingDown() /*&& (status == BT::NodeStatus::RUNNING)*/)
    {
      status = tree.tickOnce();
      std::cout<< status << std::endl;
      tree.sleep(std::chrono::milliseconds(1000));
    }
  } catch (BT::LogicError err) {
    std::cout << err.what() << std::endl;
  }
  return 0;
}

``` 

The corresponding XML can be found below:
**XML Tree**
```
<root BTCPP_format="4" >
  <BehaviorTree ID="GenericT">
    <SequenceWithMemory>
      <Script code=" hasTask:=false; LOW_BATT:=10 "/>
      <ReactiveSequence>
        <CheckLevel deviceType="BATT"
                  percentage="{LOW_BATT}"
                  isLowBattery="{isLowBattery}"/>
        <SendWarning deviceType="BATT" deviceStatus="{isLowBattery}" _skipIf="!hasTask"/> <!--should run on the second tick - NOT OK -->
        <SendWarning deviceType="545" deviceStatus="{isLowBattery}" _onSuccess="hasTask=true"/>
        <SendWarning deviceType="888" deviceStatus="{isLowBattery}" _skipIf="hasTask"/> <!-- should never run since pre-cond is satisfied - OK -->
      </ReactiveSequence>
    </SequenceWithMemory>
  </BehaviorTree>
</root>
``` 

**Observation**
1. The first instance of **SendWarning** shouldn't run on the first tick since **hasTask** is false. But should run on subsequent ticks when the pre-condition becomes true. Not working as expected!
2. The second instance of **SendWarning** is setting the pre-cond/ post-cond to true **onSuccess**. This can be proven by the the third instance of **SendWarning** which should never execute since the pre-cond (hasTask) is now set to true. Works as expected.

I hope this helps to narrow down the issue. And I apologize for the delay.

facontidavide, 2023-03-28:
I finally understood what your problem is.

You execute the **Script** in every single tick!!!! This happens because all the children of **ReactiveSequence** are synchronous, therefore, the entire **SequenceWithMemory** will execute in a single tick.

As a consequence, the next time you tick again, **Script** is executed and the very first **SendWarning** will always be skipped

ephson24, 2023-03-29:
Hi Davide, 
I still think there maybe a bug somewhere and probably the previous code example didn't do much justice in showing it.
Here are a few modifications I made to the code:
1. I have added in a new action **Initialise** which returns FAILURE **ONLY ONCE**, on the first run; causing the **Script** to be executed. It returns SUCCESS on all subsequent ticks and hence **Script** is never executed again.
2. In the XML, I have placed the **Initialise** and **Script** nodes under a **Fallback** parent.
3. According to the explanation given, the **first** instance of **SendWarning** should now be executed at some point since I am not executing **Script** in every tick.

Regardless, the behavior remains the same. The pre-condition seem not to change to reflect the later modifications made to it by the **second** instance of **SendWarning**.

```c++
#include <ros/ros.h>
#include <ros/package.h>
#include <behaviortree_cpp/behavior_tree.h>
#include <behaviortree_cpp/bt_factory.h>
#include <behaviortree_cpp/loggers/bt_cout_logger.h>
#include <boost/bind.hpp>

enum DeviceType { BATT=1, CONTROLLER=2 };
bool init = false;
BT::NodeStatus initialise()
{
  if(!init)
  {
    init = true;
    ROS_INFO("Init");
    return BT::NodeStatus::FAILURE;
  }
  return BT::NodeStatus::SUCCESS;
}

BT::NodeStatus checkLevel(BT::TreeNode &self)
{
  double percent;
  if(auto res = self.getInput<double>("percentage"))
    percent = res.value();
  else
    throw std::runtime_error(res.error());

  DeviceType devType;
  if(auto res = self.getInput<DeviceType>("deviceType"))
    devType = res.value();
  else
    throw std::runtime_error(res.error());

  if(devType == DeviceType::BATT)
  {
    self.setOutput<bool>("isLowBattery", (50 < percent));
  }

  ROS_INFO_STREAM("Device| " << devType << " Level| " << percent);
  return BT::NodeStatus::SUCCESS;
}

BT::NodeStatus sendWarning(BT::TreeNode &self)
{
  DeviceType devType;
  if(auto res = self.getInput<DeviceType>("deviceType"))
    devType = res.value();
  else
    throw std::runtime_error(res.error());
  bool devStatus = self.getInput<bool>("deviceStatus").value();
  //! Publish device warning on dedicated topic
  if(devType == DeviceType::BATT)
  {
    ROS_INFO("Processing warning update for battery ");
  }
  ROS_INFO_STREAM("Sending warning for | " << devType << " status: " << devStatus);
  return BT::NodeStatus::SUCCESS;
}

void registerNodes(BT::BehaviorTreeFactory &factory)
{
  factory.registerSimpleAction("Initialise", boost::bind(initialise));
  factory.registerSimpleCondition("CheckLevel",
                               boost::bind(checkLevel, _1),
                               { BT::InputPort<double>("percentage"),
                                 BT::InputPort<int>("deviceType"),
                                 BT::OutputPort<bool>("isLowBattery")});
  factory.registerSimpleAction("SendWarning",
                               boost::bind(sendWarning, _1),
                               { BT::InputPort<int>("deviceType"),
                                 BT::InputPort<bool>("deviceStatus")});

  factory.registerScriptingEnums<DeviceType>(); //registering ENUMS here
}

int main(int argc, char** argv)
{
  ros::init(argc, argv, "robot_bt_cpp");
  ros::NodeHandle gnh, pnh("~");

  BT::BehaviorTreeFactory factory;
  registerNodes(factory);

  std::string bt_path = ros::package::getPath("mobrob_agent");
  auto tree = factory.createTreeFromFile(bt_path + "/config/test_bt.xml");
  BT::StdCoutLogger cout_logger(tree);
  try {
    BT::NodeStatus status;
    while(not ros::isShuttingDown() /*&& (status == BT::NodeStatus::RUNNING)*/)
    {
      status = tree.tickOnce();
      std::cout<< status << std::endl;
      tree.sleep(std::chrono::milliseconds(1000));
    }
  } catch (BT::LogicError err) {
    std::cout << err.what() << std::endl;
  }
  return 0;
}

``` 

And the modified XML:

```
<root BTCPP_format="4" >
  <BehaviorTree ID="GenericT">
    <SequenceWithMemory>
      <Fallback>
        <Initialise/>
        <Script code=" hasTask:=false; LOW_BATT:=10 "/>
      </Fallback>
      <ReactiveSequence>
        <CheckLevel deviceType="BATT"
                  percentage="{LOW_BATT}"
                  isLowBattery="{isLowBattery}"/>
        <SendWarning deviceType="BATT" deviceStatus="{isLowBattery}" _skipIf="!hasTask"/> <!--should run on the second tick - NOT OK -->
        <SendWarning deviceType="545" deviceStatus="{isLowBattery}" _onSuccess="hasTask=true"/>
        <SendWarning deviceType="888" deviceStatus="{isLowBattery}" _skipIf="hasTask"/> <!-- should never run since pre-cond is satisfied - OK -->
      </ReactiveSequence>
    </SequenceWithMemory>
  </BehaviorTree>
</root>
```

ephson24, 2023-03-29:
Here is an **output from the logger**. 

```typescript
[1680095410.957]: SequenceWithMemory        IDLE -> RUNNING
[1680095410.957]: Fallback                  IDLE -> RUNNING
[1680095410.957]: Initialise                IDLE -> RUNNING
[ INFO] [1680095410.956844299]: Init
[1680095410.958]: Initialise                RUNNING -> FAILURE
[1680095410.958]: Script                    IDLE -> SUCCESS
[1680095410.958]: Initialise                FAILURE -> IDLE
[1680095410.958]: Script                    SUCCESS -> IDLE
[1680095410.958]: Fallback                  RUNNING -> SUCCESS
[1680095410.958]: ReactiveSequence          IDLE -> RUNNING
[ INFO] [1680095410.957964001]: Device| 1 Level| 10
[1680095410.958]: CheckLevel                IDLE -> SUCCESS
[1680095410.958]: SendWarning               IDLE -> IDLE
[1680095410.958]: SendWarning               IDLE -> IDLE
[1680095410.958]: SendWarning               IDLE -> RUNNING
[ INFO] [1680095410.958056419]: Sending warning for | 545 status: 0
[1680095410.958]: SendWarning               RUNNING -> SUCCESS
[1680095410.958]: SendWarning               IDLE -> IDLE
[1680095410.958]: SendWarning               IDLE -> IDLE
[1680095410.958]: CheckLevel                SUCCESS -> IDLE
[1680095410.958]: SendWarning               SUCCESS -> IDLE
[1680095410.958]: ReactiveSequence          RUNNING -> SUCCESS
[1680095410.958]: Fallback                  SUCCESS -> IDLE
[1680095410.958]: ReactiveSequence          SUCCESS -> IDLE
[1680095410.958]: SequenceWithMemory        RUNNING -> SUCCESS
[1680095410.958]: SequenceWithMemory        SUCCESS -> IDLE
SUCCESS
[1680095411.958]: SequenceWithMemory        IDLE -> RUNNING
[1680095411.958]: Fallback                  IDLE -> RUNNING
[1680095411.958]: Initialise                IDLE -> RUNNING
[1680095411.958]: Initialise                RUNNING -> SUCCESS
[1680095411.958]: Initialise                SUCCESS -> IDLE
[1680095411.958]: Fallback                  RUNNING -> SUCCESS
[1680095411.959]: ReactiveSequence          IDLE -> RUNNING
[ INFO] [1680095411.958661460]: Device| 1 Level| 10
[1680095411.959]: CheckLevel                IDLE -> SUCCESS
[1680095411.959]: SendWarning               IDLE -> RUNNING
[ INFO] [1680095411.958907648]: Processing warning update for battery 
[ INFO] [1680095411.958991565]: Sending warning for | 1 status: 0
[1680095411.959]: SendWarning               RUNNING -> SUCCESS
[1680095411.959]: SendWarning               IDLE -> RUNNING
[ INFO] [1680095411.959165563]: Sending warning for | 545 status: 0
[1680095411.959]: SendWarning               RUNNING -> SUCCESS
[1680095411.959]: SendWarning               IDLE -> IDLE
[1680095411.959]: SendWarning               IDLE -> IDLE
[1680095411.959]: CheckLevel                SUCCESS -> IDLE
[1680095411.959]: SendWarning               SUCCESS -> IDLE
[1680095411.959]: SendWarning               SUCCESS -> IDLE
[1680095411.959]: ReactiveSequence          RUNNING -> SUCCESS
[1680095411.960]: Fallback                  SUCCESS -> IDLE
[1680095411.960]: ReactiveSequence          SUCCESS -> IDLE
[1680095411.960]: SequenceWithMemory        RUNNING -> SUCCESS
[1680095411.960]: SequenceWithMemory        SUCCESS -> IDLE
SUCCESS

```

facontidavide, 2023-05-20:
Problem fixed in branch v4.2, as far as I can tell

closed 2023-05-20 by facontidavide