BehaviorTree/BehaviorTree.CPP #538: Unable to build with project

ephson24, 2023-03-31:
Hi Davide, 
I am unable to compile my project since the recent commit. I get this error that says: 
```c++
/home/ephson/mobrob_ws/src/BehaviorTree.CPP/include/behaviortree_cpp/behavior_tree.h:-1: In function 'const char* BT::LibraryVersionString()':
/home/ephson/mobrob_ws/src/BehaviorTree.CPP/include/behaviortree_cpp/behavior_tree.h:95: error: 'BTCPP_LIBRARY_VERSION' was not declared in this scope
   95 |   return BTCPP_LIBRARY_VERSION;
      |          ^~~~~~~~~~~~~~~~~~~~~
``` 
Is this something you can take a look at?

Ale-Amici, 2023-03-31:
I have experience the same issue compiling with ROS1 using 'catkin build'

Ale-Amici, 2023-03-31:
I have solved it adding
`#define BTCPP_LIBRARY_VERSION "@BTCPP_LIBRARY_VERSION@"
`
in that file that produces the error. Now, the variable `BTCPP_LIBRARY_VERSION` is created and it gets the value from the CMake command
`target_compile_definitions(${BTCPP_LIBRARY} PUBLIC BTCPP_LIBRARY_VERSION="${CMAKE_PROJECT_VERSION}")
`

Ale-Amici, 2023-03-31:
no, in my case everything was alright :sweat_smile: 

It might be that different compilers, or building methods treat those compile definitions differently :shrug: 

With ROS1 catkin build, everything is fine with the fix I suggested. But maybe it's not necessary with other build methods?

closed 2023-03-31 by facontidavide

Ale-Amici, 2023-03-31:
yes! This fixed it! thanks :)