BehaviorTree/BehaviorTree.CPP #539: Arithmetic operations on Post Conditions (Success | Failure) | Not working

ephson24, 2023-03-31:
Hi Davide,

I find trying to set arithmetic operations using **_onSuccess** and **_onFailure** does not work.
However using **_onpost** works as expected.

E.g XML (not working)
```
<Sequence>
              <TestCanDo state="false"
                         _onFailure="loop_count  += 1"/>
              <SendWarning deviceStatus="{isLowBattery}"
                           deviceType="BATT"/>
</Sequence>
``` 
But this works:
```
<Sequence>
              <TestCanDo state="false"
                         _onpost="loop_count  += 1"/>
              <SendWarning deviceStatus="{isLowBattery}"
                           deviceType="BATT"/>
</Sequence>
``` 

Could this be a bug of sorts?

PS: **TestCanDo** is a simple condition node

ephson24, 2023-04-03:
Solved by removing the spacing between the variable and arithmetic operations
**Solution**
```
<Sequence>
              <TestCanDo state="false"
                         _onFailure="loop_count+=1"/>
              <SendWarning deviceStatus="{isLowBattery}"
                           deviceType="BATT"/>
</Sequence>
```

facontidavide, 2023-04-03:
That is still a bug, even with spacing, it should work.

facontidavide, 2023-05-10:
I added a specific unit test for your case, but can not reproduce the error:

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/tests/gtest_postconditions.cpp#L41-L62

Closing. If youcan create a variant of the unit test that show the issue, let me know

closed 2023-05-10 by facontidavide