BehaviorTree/BehaviorTree.CPP #544: Unable to build 4.1.1 using Conan

sjpritchard, 2023-04-11:
I'm trying to build on Windows 11 using VS2022 and it throws the following erorr on the "cmake --build . --parallel" build step:

C:\Users\steve\Downloads\BehaviorTree.CPP-4.1.1\BehaviorTree.CPP\3rdparty\cpp-sqlite/sqlite.hpp(24,10
): fatal  error C1083: Cannot open include file: 'sqlite3.h': No such file or directory (compiling so
urce file C:\Users\steve\Downloads\BehaviorTree.CPP-4.1.1\BehaviorTree.CPP\src\loggers\bt_sqlite_logg
er.cpp) [C:\Users\steve\Downloads\BehaviorTree.CPP-4.1.1\build\behaviortree_cpp.vcxproj]
C:\Users\steve\Downloads\BehaviorTree.CPP-4.1.1\BehaviorTree.CPP\3rdparty\cppzmq/zmq.hpp(36,10): fata
l  error C1083: Cannot open include file: 'zmq.h': No such file or directory (compiling source file C
:\Users\steve\Downloads\BehaviorTree.CPP-4.1.1\BehaviorTree.CPP\src\loggers\groot2_publisher.cpp) [C:
\Users\steve\Downloads\BehaviorTree.CPP-4.1.1\build\behaviortree_cpp.vcxproj]

zmq and sqlite have been downloaded and installed under the .conan2 folder in my home folder, so the issue is with the ability to find those

MatteoAlbi, 2023-04-12:
I`m having the same problem on Windows 10 building with conan 1.59, cmake version 3.25.2.
In file conan_toolchain.cmake the packages paths of zmq.h and sqlite3.h are correctly defined (CMAKE_INCLUDE_PATH). 
However, paths defined in CMAKE_PROGRAM_PATH do not exist.

facontidavide, 2023-04-27:
Thankd for reporting. I will look into this this weekend

aiswaryame94, 2023-08-01:
Do we have a solution for this yet? I too ran into the same issue.

facontidavide, 2023-08-01:
Sorry about that. Unfortunately this issues is not too high on my list of priorities.

Also, I will drop any support for conan 1.x and focus on conan 2.x.

nevalsar, 2023-08-10:
I am also able to duplicate this issue with conan v1.58, but the build fails with error finding zmq (not sqlite3).
> In file included from /src/resources/bt-tree-build/BehaviorTree.CPP-4.3.4/src/loggers/groot2_publisher.cpp:6:
/src/resources/bt-tree-build/BehaviorTree.CPP-4.3.4/3rdparty/cppzmq/zmq.hpp:36:10: fatal error: zmq.h: No such file or directory.

wiped1, 2023-08-29:
Building in `Release` mode solves missing `sqlite` and `zmq` for me.

`cmake --build . --config Release --parallel`

VS2022, win10, conan 2.0.10, BehaviorTree master branch as of today

aiswaryame94, 2023-08-29:
Thank you for your response. This solved my problem too.