BehaviorTree/BehaviorTree.CPP #555: C2589 and C3861 occurred during building x86-windows

MonicaLiu0311, 2023-05-06:
Two errors occurred when building for `x86-windows` as follows:
1. Macros about `min` and `max`:
 ```
  E:\vcpkg01\buildtrees\behaviortree-cpp\src\haviorTree-00e1cfa921.clean\include\behaviortree_cpp\utils\simple_string.hpp(128): error C2589: '(': illegal token on right side of '::'
  E:\vcpkg01\buildtrees\behaviortree-cpp\src\haviorTree-00e1cfa921.clean\include\behaviortree_cpp\utils\simple_string.hpp(128): error C2062: type 'unknown-type' unexpected
  E:\vcpkg01\buildtrees\behaviortree-cpp\src\haviorTree-00e1cfa921.clean\include\behaviortree_cpp\utils\simple_string.hpp(128): error C2789: 'SafeAny::SimpleString::CAPACITY': an object of const-qualified type must be initialized
  E:\vcpkg01\buildtrees\behaviortree-cpp\src\haviorTree-00e1cfa921.clean\include\behaviortree_cpp\utils\simple_string.hpp(128): error C2059: syntax error: ')'
  E:\vcpkg01\buildtrees\behaviortree-cpp\src\haviorTree-00e1cfa921.clean\include\behaviortree_cpp\utils\simple_string.hpp(136): error C2131: expression did not evaluate to a constant
  E:\vcpkg01\buildtrees\behaviortree-cpp\src\haviorTree-00e1cfa921.clean\include\behaviortree_cpp\utils\simple_string.hpp(157): error C3863: array type 'char [CAPACITY]' is not assignable
  E:\vcpkg01\buildtrees\behaviortree-cpp\src\haviorTree-00e1cfa921.clean\include\behaviortree_cpp/utils/convert_impl.hpp(68): error C2760: syntax error: '(' was unexpected here; expected ')'
  E:\vcpkg01\buildtrees\behaviortree-cpp\src\haviorTree-00e1cfa921.clean\include\behaviortree_cpp/utils/convert_impl.hpp(68): error C3878: syntax error: unexpected token '(' following 'expression'
  ```

2. Use of functions `_BitScanForwar` and `_BitScanForward64`:
  ```
  E:\vcpkg01\buildtrees\behaviortree-cpp\src\haviorTree-00e1cfa921.clean\3rdparty\lexy\include\lexy/_detail/swar.hpp(144): error C3861: '_BitScanForward64': identifier not found
  ```

Reproduce environment:
* Win 10
* VS 2022

MonicaLiu0311, 2023-05-10:
Thanks for your timely response. If you agree, I will fix this issue in vcpkg by patch and remove it after the next version upgrade.

MonicaLiu0311, 2023-05-10:
> just pushed a fix to solve the problem in **simple_string.hpp**

There is also `convert_impl.hpp` that needs to be fixed, all the places involving `max` and `min` macros.

MonicaLiu0311, 2023-05-10:
> feel free to send a PR to the main BT.CPP repo, if that fix the issue

All right!

facontidavide, 2023-10-16:
According to CI, compiles correctly on Windows. Closing

closed 2023-10-16 by facontidavide