BehaviorTree/BehaviorTree.CPP #556: BT.CPP 4.0.1 Unexpected infinite loop using ReactiveSequence & Precondition not working on Sequence node

MatteoAlbi, 2023-05-08:
I'm trying to run the following tree:

```
<root BTCPP_format="4" >
        <BehaviorTree ID="MainTree">
            <Sequence name="init_seq">
                <Script name="init"
                        code="  x:=30; \
                                err:=0; \
                                obj_flag:=0"
                />
                <ReactiveSequence name="execution">
                    
                    <Sequence name="err_seq" _successIf="!err">
                        <ReadString input="    err triggered" />
                        <Script code="  x+=2; \
                                        err=0" />
                        <ReadDouble input="{x}" />
                    </Sequence>

                    <Fallback name="action">
                        <CondBool   input="{obj_flag}"
                                    val="1"
                        />
                        <Script code="  x-=2 " />
                        <ReadDouble input="{x}" />
                    </Fallback>

                </ReactiveSequence>
            </Sequence>
        </BehaviorTree>
    </root>
```

ReadDouble and ReadString are SyncActionNodes which print the passed value to terminal using cout. CondBool is a ConditionNode which returns SUCCESS if input is equal to val.

However, the tree does not behave as expected:
- It enters the Sequence "err_seq"
- The ReactiveSequence "execution" generates an infinite loop

facontidavide, 2023-05-08:
I am not sure what you mean by "infinite loop".

What I expect to happen is:

- `obj_flag` is always 0 and the `CondBool` always return FAILURE.
- ` ReadDouble` inside the `Fallback` is NEVER executed, because the second child, `Script` return SUCCESS.
- At each tick of the `ReactiveSequence`  we will both increment and decrement X

How are you ticking the tree? If you share the code with me, I will try to debug this today.

MatteoAlbi, 2023-05-08:
Indeed I should have put the last `ReadDouble` outside the fallback node. 
However, I would not expect X to decrement because the precondition `_successIf="!err"` should skip the sequence node `err_seq`.
Anyway, what happens is that the Sequence `err_seq` is run indefinetly, and the tree never accesses the fallback node `action`

facontidavide, 2023-05-08:
I think that I found the bug, and it is a pretty serious on in `ReactiveSequence`.
I will try to come back to you with a solution as soon as possible

facontidavide, 2023-05-10:
that works too, once the other problem is solved.

Can you please try branch **v4.2** ?

There is a unit test that reproduce your error: https://github.com/BehaviorTree/BehaviorTree.CPP/blob/v4.2/tests/gtest_reactive.cpp

The problem is that **ReactiveSequence** can not have multiple children that return RUNNING, but because of the changes I describe [here](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/v4.2/tests/gtest_reactive.cpp), 
now the first Sequence is asynchronous.

This apparently was a bad decision and I am reverting the default behavior in version 4.2

MatteoAlbi, 2023-05-12:
I compiled version 4.2 and all tests are successful, included Reactive.RunningChildren

closed 2023-05-19 by MatteoAlbi