BehaviorTree/BehaviorTree.CPP #557: tickOnce not working?

fwollatz, 2023-05-09:
In this [tutorial ]( https://www.behaviortree.dev/docs/tutorial-basics/tutorial_04_sequence)you use the tickOnce function, to loop through the tree, instead of the tree.tickWhileRunning(). Confusingly my tree seems to be acting exactly the same wether or not I use tickOnce or tickWhileRunning. Is this a bug, or am I doing something wrong?

The only real difference between your code and mine is, that yours uses StatefulActionNodes, while mine consists of SyncActionNode. If this is the case, please close this (and maybe add a comment in the documentation or at least in this thread)

facontidavide, 2023-05-09:
No, it is not an error, if ALL your nodes are synchronous.

You will see a difference only if any of your node is asynchronous

fwollatz, 2023-05-09:
Ok, thanks! :D

closed 2023-05-09 by facontidavide