BehaviorTree/BehaviorTree.CPP #563: Issue with ports not being remapped with subtreeplus on v3.8

Rid0n, 2023-05-12:
Here's my code. 

I have the outermost tree passing down values to a subtree which is then passing down values to another subtree which ticks a navigation tree. Everything with the `__autoremap` flag set to true. Very simple.

    <root main_tree_to_execute="Tree1">
    <include path="./navigation_tree.xml"/>

    <BehaviorTree ID="Tree1">

        <Sequence>

            <SubTreePlus ID="Tree2" __autoremap="true"/>

        </Sequence>

    </BehaviorTree>

    <BehaviorTree ID="Tree2">

        <Sequence>

            <SubTreePlus ID="Tree3" __autoremap="true"/>

        </Sequence>

    </BehaviorTree>

    <BehaviorTree ID="Tree3">

        <Sequence>

            <SubTreePlus ID="NavigateTree" __autoremap="true"/>

        </Sequence>

    </BehaviorTree>

    </root>

I expect there not to be any issues, however for some reason I'm getting an error

`[bt_navigator]: Exception when loading BT: Blackboard::get() error. Missing key [node]`

I found that the issue lies specifically in the switch from Tree2 to Tree3. Every other subtree node does successfully remap everything.

If I change one line
`<SubTreePlus ID="Tree3" __autoremap="true"/> `
To
`<SubTreePlus ID="Tree3" goal="{goal}" node="{node}" bt_loop_duration="{bt_loop_duration}" tf_buffer="{tf_buffer}" server_timeout="{server_timeout}" />`
It all works fine.

What's up here?

facontidavide, 2023-06-10:
Version 4.X has been in development for almost 1 year now. Keep maintaining 3.8 is really a struggle for me.

Said that, would you be so kind to give me a unit test that I can use to reproduce the issue?

facontidavide, 2023-06-10:
These are my current tests: https://github.com/BehaviorTree/BehaviorTree.CPP/blob/v3.8/tests/gtest_subtree.cpp

facontidavide, 2023-06-10:
I was able to reproduce here: https://github.com/BehaviorTree/BehaviorTree.CPP/tree/issue563

I will keep you posted

facontidavide, 2023-06-12:
Hi, problem fixed in branch **issue563**.

Would you be so kind to test it?

Once you confirm, I will merge and release a new version

Rid0n, 2023-06-13:
still there for me D: 
it also broke the delicate ways in which it did work haha

trying to find a test case where that'll show

facontidavide, 2023-06-13:
oh god, don't tell me that!

Rid0n, 2023-06-20:
So, the unit test passes fine, but when I'm trying a simple thing with the nav2 stack, it fails. 

    <root main_tree_to_execute="Tree">

    <BehaviorTree ID="Tree">

        <Sequence>

            <SubTreePlus ID="Main" __autoremap="true"/>

        </Sequence>

    </BehaviorTree>

    <BehaviorTree ID="Main">

        <Wait wait_duration="1"/>

    </BehaviorTree>

    </root>

fails to instantiate tree with 
`Blackboard::get() error. Missing key [node]`

the `bt_navigator` node crashes entirely, so it looks like there's something nasty together with nav2 specifically.

facontidavide, 2023-06-20:
Sorry, I just implemented it and it works.

I don't know how to help you sorry.

```c++
class WaitNode : public BT::StatefulActionNode
{
  std::chrono::system_clock::time_point _deadline;
public:
  WaitNode(const std::string& name, const BT::NodeConfiguration& config) :
    BT::StatefulActionNode(name, config)
  {}

  virtual NodeStatus onStart() override
  {
    double time = getInput<double>("wait_duration").value();
    _deadline =
        std::chrono::system_clock::now() + std::chrono::milliseconds(int(time * 1000));
    return NodeStatus::RUNNING;
  }

  virtual NodeStatus onRunning() override
  {
    return (std::chrono::system_clock::now() < _deadline) ? NodeStatus::RUNNING :
                                                            NodeStatus::SUCCESS;
  }

  virtual void onHalted() override {}

  static BT::PortsList providedPorts()
  {
    return {BT::InputPort<double>("wait_duration")};
  }
};

TEST(SubTree, Autoremap)
{
  BehaviorTreeFactory factory;
  const std::string tree_xml = R"(
<root main_tree_to_execute="Tree">
  <BehaviorTree ID="Tree">
    <Sequence>
        <SubTreePlus ID="Main" __autoremap="true"/>
    </Sequence>
  </BehaviorTree>

  <BehaviorTree ID="Main">
    <Wait wait_duration="1"/>
  </BehaviorTree>
</root>
)";

  factory.registerNodeType<WaitNode>("Wait");
  Tree tree = factory.createTreeFromText(tree_xml);
  auto ret = tree.tickRootWhileRunning();
  ASSERT_EQ(ret, NodeStatus::SUCCESS);
}
```

facontidavide, 2023-06-20:
Also the message: 

`Blackboard::get() error. Missing key [node]`

suggests that you are having a problem with the infamous use of the blackboard to share the **ROS node handle**.

This is a bad practice and an abuse of the blackboard that I don't support, nor recommend.

I think the solution you need has nothing to do with SubtreePlus and autoremap.

Rid0n, 2023-06-20:
it is the way it's currently done all over nav2 bt's, unfortunately.
you mentioned this issue in a nav2 thread asking if it breaks something - it does.

Rid0n, 2023-06-20:
it does look like you fixed it from your side, but that breaks the nav2 side somehow

facontidavide, 2023-06-20:
I don't even know how it is possible that it worked before.

Also, I think that, as you detected it yourself (motivation of this issue),  `Subtree / __shared_blackboard` works, but `SubtreePlus / __autoremap`  does NOT work.

Is that a correct statement? This make sense to me considering what Nav2 does and the way SubtreePlus autoreamp was implemented.

In other words, unless we change something in Nav2 (I have a suggestion), `SubtreePlus / __autoremap` did not work as YOU expect (wrong expectation) before and it will not work now.

Said that, I DO NEED to know what used to work but is not working anymore.

Rid0n, 2023-06-20:
I wanted to use `SubtreePlus / __autoremap` because that would use a separate context and not impact the parent one. A one way data flow. (if I understood the implementation correctly anyway)

I was using `SubtreePlus` and for some subtrees, `__autoremap` worked(!!), but for others it didn't(??), I never figured out what that depended on, it seemed random. This is when I created the issue here.

facontidavide, 2023-06-20:
I have no idea how it could possibly work before

facontidavide, 2023-06-20:
I have an idea, maybe it is possible to fix this, I will ask you later to try a new branch

Rid0n, 2023-06-20:
fun to be a part of ! haha

facontidavide, 2023-06-20:
Please try this branch **issue_563**

Rid0n, 2023-06-20:
> Please try this branch **issue_563**

nope ;(
Same error...
`bt_navigator` still crashes..

facontidavide, 2023-06-20:
I will need to try this on my machine or debug this with you. Can you arrange a session tomorrow or Thursday to do the investigation together?

facontidavide, 2023-06-21:
Hi.

any additional detail? When you say "crashes" you mean that an exception in thrown? Same message?

Rid0n, 2023-06-21:
sorry, don’t have time now. 
I think it segfaults. the ros node crashes. 
cant you get the nav2_behavior_tree package and debug using its unit tests?

facontidavide, 2023-06-21:
> cant you get the nav2_behavior_tree package and debug using its unit tests?

Honestly, fixing what Nav2 does is not on top of my priorities; I will prioritize what can be reproduced in **my** unit tests.
I might have a look at that over the weekeend...

SteveMacenski, 2023-06-21:
Nor should you, @Rid0n its on you to provide reproduction help or compile with a debugger to get a trace. Its not @facontidavide 's problem to go into another project he doesn't maintain to chase down your issue :smile:

facontidavide, 2023-06-21:
@Rid0n to be clear: I am grateful for the time you spent so far. I was able to correct some issues and improve the library thanks to your error report.

But I need more information to help you further.

Rid0n, 2023-06-21:
> Nor should you, @Rid0n its on you to provide reproduction help or compile with a debugger to get a trace. Its not @facontidavide 's problem to go into another project he doesn't maintain to chase down your issue 😄
> 
> You should provide the level of detail necessary for a responsible maintainer to make actionable progress on, not just "nope ;("

I know, but isn’t it a bit of a gray area since it’s something that’s affecting both projects?

I only said that because @facontidavide seems to want to fix this quickly and without fallout to nav2, while I can’t quickly provide what he’s asking for. ;(

Rid0n, 2023-06-21:
> @Rid0n to be clear: I am grateful for the time you spent so far. I was able to correct some issues and improve the library thanks to your error report.
> 
> But I need more information to help you further.

and I will give it to you if I find the time haha 💌

facontidavide, 2023-07-21:
as far as I know, this is fixed

m-morelli, 2023-07-31:
Ciao @facontidavide, same problem as @HovorunB here. I've recently upgraded my `ros-humble-navigation2` stack from v1.1.6 to v1.1.8, which in turn upgraded `ros-humble-behaviortree-cpp-v3` from v3.8.3 to v3.8.4. Now when my BT executor executes a `SubTreePlus` like
```
<SubTreePlus name="Patrol" ID="Patrol" tl1="KITCHEN" tl2="LIVING"/>
```
I get this error:
```
[ERROR] [1690826385.151509220] []: Original error: The creation of the tree failed because the port [tl1] was initially created with type [std::string] and, later type [std_msgs::msg::String_<std::allocator<void> >] was used somewhere else.
```

_It was used to work in the previous version_. Possibly related to https://github.com/BehaviorTree/BehaviorTree.CPP/issues/489, which you already fixed for BT.CPP v4?

facontidavide, 2023-07-31:
I am spending some days with my family with little access to the computer.

If anyone can provide a simple example that I can reproduce locally, I will focus my attention to this issue as soon as I can

facontidavide, 2023-08-01:
@m-morelli would you be so kind to open a new issue and provide a way to reproduce it?

m-morelli, 2023-08-01:
Ciao @facontidavide , sure I'll do it as soon as I have a bit of time. Thanks!

m-morelli, 2023-08-01:
Done #623. Thanks!

tonynajjar, 2023-08-16:
This can be closed then I guess

closed 2023-08-16 by facontidavide