BehaviorTree/BehaviorTree.CPP #567: Initial compilation process with Conan not working on Windows

NKRVVI, 2023-05-21:
The third line of the initial compilation process is returning errors for me on my Windows machine : cmake ../BehaviorTree.CPP -DCMAKE_TOOLCHAIN_FILE=conan_toolchain.cmake

The error that I am getting is : "CMake Error: The source directory "C:/BehaviorTree.CPP_parent/build/.cmake" does not exist."

If i remove the ".cmake" from the command, then i get this error : 
"Could not find toolchain file: conan_toolchain
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!"

facontidavide, 2023-10-16:
as long as it works in CI, I will consider this to be a problem with your configuration

skhaz, 2023-10-29:
I am having the issue, on my machine it works (macOS), on CI not. 

My steps:

```

- name: Checkout
  uses: actions/checkout@v4

- name: Install Conan
  run: |
    pip install conan

- name: Prepare build directory
  run: mkdir build

- name: Install Dependencies
  working-directory: build
  run: |
    conan profile detect
    conan install ..  --output-folder=. --build=missing --settings compiler.cppstd=20 --conf tools.system.package_manager:mode=install

- name: Configure
  working-directory: build
  run: |
    ls
    cmake .. -DCMAKE_TOOLCHAIN_FILE=conan_toolchain.cmake -DCMAKE_BUILD_TYPE=Release -G "Visual Studio 17 2022" -A x64

- name: Build
  working-directory: build
  run: |
    cmake --build . -j $(nproc)
```

I am using `windows-2022` on GitHub Actions.

facontidavide, 2023-10-30:
I will have a look, thanks

skhaz, 2023-10-30:
@facontidavide, I have found the error, is it because the Windows' shell splits the dot, we have to pass the conan toolchain between quotes, like this "conan_toolchain.cmake".

closed 2023-11-03 by facontidavide