BehaviorTree/BehaviorTree.CPP #569: Question: Type-safety of the blackboard values

MithunKinarullathil, 2023-05-26:
Hi,

From the [documentation](https://www.behaviortree.dev/docs/tutorial-basics/tutorial_02_basic_ports/) I learned that the values in the blackboard are type-safe. Further reading I saw this section
```
BehaviorTree.CPP supports automatically converting strings into common types, such as int, long, double, bool, NodeStatus, etc. User-defined types can be supported easily as well.
```
My doubt this: are the values in the blackboard always stored as strings? If so, is there a way to guarantee type safety? 

eg: Image a key:value pair with std:int type and value 3. In another action node, later on in the tree, would I be able to save an std:float into the same key?

Thank you for your time.

facontidavide, 2023-05-26:
great question!

# BB storage

Storage is done with a variant of [std::any](https://en.cppreference.com/w/cpp/utility/any). if you are unfamiliar with it, I strongly recommend you learn how that works.
**std::any** implements  type erasure (store any type), but preserving the original object AND its type.

NOW, in addition to any, we do:
- **small object optimization** to prevent allocating memory in the heap if the object is 16 bytes or less.
- **smart casting**. For instance, a regular `std::any` would complain if you try to cast `uint16_t` to `int32_t`, but instead I check if there is any numberical error in the casting, and if there isn't, it is allowed.

But wait, there is more.

If you DO KNOW the type of a blackboard entry / port, we will automatically TRY to convert the string to that type. 

## Answer

I am assuming you will use the provided `getInput` and `setOutput` interfaces (as you should).

You can **not** connect a `OutputPort<float>` to `InputPort<int>` because they haven't the same type.

If everything works as intended, you should get an exception during the creation of the tree.

You can do, anyway `getInput<float>()` of an integer port, if that is not causing numerical cancellation.

closed 2023-05-26 by facontidavide

MithunKinarullathil, 2023-05-26:
@facontidavide Great, thanks a lot for clearing that up.