BehaviorTree/BehaviorTree.CPP #573: what():  [ReactiveSequence]: only a single child can return RUNNING

Ale-Amici, 2023-06-02:
Hello :) 
In my workspace, I updated the BehaviorTree.CPP library from version 4.0.2 to 4.2.0 and this error appeared when using ReactiveSequence:
`what():  [ReactiveSequence]: only a single child can return RUNNING`

This is the tree:
```
<ReactiveSequence name="LocalizationSequence">
    <CheckSensorQuality>
        <RetryUntilSuccessful name="SensorRecoveryLoop" num_attempts="5">
            <RestartSensor/>
        </RetryUntilSuccessful>
    </CheckPoseQuality>
    <PerformActionA/>
</ReactiveSequence>
```

We have a node `CheckSensorQuality` that checks a value and a node `PerformActionA`.They are two children of a ReactiveSequence. 
`CheckSensorQuality` returns SUCCESS unless some sensor data is bad, then it returns RUNNING and it starts its child `SensorRecoveryLoop` (which is a `RetryUntilSuccessful` node) that has a child `RestartSensor`.
`PerformActionA` return always RUNNING.

The expected behavior is that `CheckSensorQuality` going to RUNNING should halt `PerformActionA`. This seems to happen, but then the tree crashes with the above-mentioned error. What could I have done wrong in the logic of my nodes? I didn't get this error with version 4.0.2.

`[1685687135.585]: ReactiveSequence      IDLE -> RUNNING
[1685687135.585]: CheckSensorQuality          IDLE -> SUCCESS
[1685687135.585]: PerformActionA      IDLE -> RUNNING

[1685687157.844]: CheckPoseQuality          SUCCESS -> RUNNING
[1685687157.844]: SensorRecoveryLoop  IDLE -> RUNNING
[1685687157.844]: RestartSensor       IDLE -> RUNNING
[1685687157.845]: PerformActionA      RUNNING -> IDLE
terminate called after throwing an instance of 'BT::LogicError'
  what():  [ReactiveSequence]: only a single child can return RUNNING
================================================================================REQUIRED process [sequential_control-1] has died!`

Thank you in advance for your time and support!

facontidavide, 2023-06-06:
Yes, what you are trying to do is broken.

`ReactiveSequence`  works only if, given N children, the first N-1 are fully synchronous and the last is asynchronous.

You should reconsider your logic taking that into account

closed 2023-06-06 by facontidavide

Ale-Amici, 2023-06-06:
I see, thank you very much, could more information about this be added in the documentation website? I think we read somewhere that if the another child returned running the others were halted, but as you said there can only be one asynchronous node :sweat_smile: 

Thank you very much for your support

facontidavide, 2023-06-06:
The Parallel node is the only one that can have multiple children in RUNNING state. 
That deserves its own tutorial, eventually