BehaviorTree/BehaviorTree.CPP #574: Robotic Locomotion BehaviourTree

dhruvthanki, 2023-06-03:
Hello!
I am a grad student working on bipedal robotic locomotion.

I have three controllers: Left stance, right stance, and double stance controllers. I am trying to implement a BT class that switches the controller based on the robot's state as input and generates joint torques using something like what's shown below. The idea is to start with a double stance phase and wait for the input from the user to change to either a left or right stance and then keep switching between any of the three based on the state of the root. And then transition to double stance and stop based on user input.
```cpp
// bt_WalkingController.h
#include <WholeBodyController.h>

class bt_WalkingController {
       // required BT members

        WholeBodyController<WholeBodyControllerStanceType::LEFT> wbc_left;
        WholeBodyController<WholeBodyControllerStanceType::RIGHT> wbc_right;
        WholeBodyController<WholeBodyControllerStanceType::DOUBLE> wbc_double;
        
        State state;
        Torques torques;

        void set_state(State state);
        void get_torques(Torques torques);
        void update();
};
```
What does the structure of the class and the update method look like? I don't have experience working with BTs. Any advice on what type of formulation to use and how to implement this using the library would be helpful.

Thank you in advance.
Dhruv

SteveMacenski, 2023-06-03:
You should read the first 4 chapters of this book to get a grasp: https://arxiv.org/abs/1709.00084

facontidavide, 2023-06-06:
Funny enough, I worked on biped locomotion for 5 years.

My two cents... don't use behavior trees to do that. Just do a very simple sequencer / state machine.

closed 2023-06-06 by facontidavide