BehaviorTree/BehaviorTree.CPP #580: Default values for ports with generic types

OPerepadia, 2023-06-09:
I would like to set default values for ports with generic types.

This code from tutorial 3 compiles and works as expected:
```cpp
return {InputPort<Position2D>("target", description)};
```

But this one produces an error - no matching function for call to ‘to_string(BT::Position2D&)’
```cpp
return {InputPort<Position2D>("target", {1.0, 2.0}, description)};
```

Is this not implemented or am I missing something in my code?

OPerepadia, 2023-06-09:
Solved. I added `toStr` function so it knows how to convert this type back to string, and now it compiles and works as expected.

```cpp
template <>
inline std::string toStr<Position2D>(Position2D value) {
    std::ostringstream ss;
    ss << value.x << ";" << value.y;
    return ss.str();
}
```

OPerepadia, 2023-06-09:
@facontidavide Maybe we should include this function in the tutorial?

facontidavide, 2023-06-09:
interesting, I would say that this is **almost** a bug. Let me add a more useful error message

facontidavide, 2023-06-09:
changed the message error to be more useful

closed 2023-06-09 by facontidavide

facontidavide, 2023-06-10:
Sooooo...
The solution to this problem was finally more complicated that I expected.

I have a new branch v4.3, where I implemented a considerable improvement of the default value for custom types.

You may try that, if you want.

commit 77c0571e (2023-06-09): Fix #580 : more informative error when not specializing BT::toStr