BehaviorTree/BehaviorTree.CPP #586: BT.CPP 4.3.0 Reactive sequences 'multiple children RUNNING' error

kd-ufr, 2023-06-14:
Hi Davide,

I think I've found a bug where reactive sequence incorrectly believes it has multiple RUNNING children and throws an error. I've included an example below.

```
  <BehaviorTree ID="Example A">
    <Sequence>
      <Script code="number := 0"/>
      <ReactiveSequence>
        <RetryUntilSuccessful name="Retry 1"
                              num_attempts="-1">
          <AlwaysFailure name="Failure 1"
                         _successIf="number >= 1"
                         _onFailure="number = number + 1"/>
        </RetryUntilSuccessful>
        <RetryUntilSuccessful name="Retry 2"
                              num_attempts="-1">
          <AlwaysFailure name="Failure 2"/>
        </RetryUntilSuccessful>
      </ReactiveSequence>
    </Sequence>
  </BehaviorTree>
```

I expect the tree to tick Retry 1 and Failure 1 twice. On the second tick both SUCCEED and the tree ticks Retry 2 and Failure 2. However, after ticking Retry 2 and Failure 2, the tree fails.

```
Sequence                  IDLE -> RUNNING
Script                    IDLE -> SUCCESS
ReactiveSequence          IDLE -> RUNNING
Retry 1                   IDLE -> RUNNING
Failure 1                 IDLE -> FAILURE
Failure 1                 FAILURE -> IDLE
Failure 1                 IDLE -> SUCCESS
Failure 1                 SUCCESS -> IDLE
Retry 1                   RUNNING -> SUCCESS
Retry 2                   IDLE -> RUNNING
Failure 2                 IDLE -> FAILURE
Failure 2                 FAILURE -> IDLE
terminate called after throwing an instance of 'BT::LogicError'
   what():  [ReactiveSequence]: only a single child can return RUNNING

```

I believe this is caused in `reactive_sequence.cpp`, where if a child node is running, `running_child_` is set to its index. However, when the child node succeeds, `running_child_` is not reset to -1. This means when a second child returns running, the reactive sequence thinks two children are running and errors. I've tried resetting `running_child_` on a child success if the index matches, and the error no longer occurs. Not sure if this would cause knock on effects though?

Thanks for your help!

facontidavide, 2023-06-14:
Yes, I added that exception to remind you that you are doing something that you are not supposed to.

`ReactiveSequence` is supposed to be used with N-1 children that are synchronous and only one that is asynchronous,
and will be interrupted if the any of the others return FAILURE.

The problem in your case is that `RetryUntilSuccessful` is potentially an **asynchronous**  node.

Having the dummy code here is very useful (thanks for simplifying it for me) but can you describe what you are trying to do? I have the feeling that from a logical point of view, you may not want to use **ReactiveSequence**.

kd-ufr, 2023-06-14:
Thanks for getting back to me. I have a reactive sequence that is trying to capture this behavior:
- update a variable in the blackboard every tick
- wait until the variable is true
- once true, proceed with remaining children while checking variable remains true
- if variable goes false (after having been true), reactive sequence fails

This has resulted in a tree snippet that looks like this:
```
    <ReactiveSequence>
      <VariableUpdaterConditionNode topic_name="/test_topic"
                                 vehicle="{variable}"/>
      <RetryUntilSuccessful name="Retry 1"
                            num_attempts="-1"
                            _skipIf="variable ">
        <AlwaysFailure name="Failure 1"
                       _successIf="variable"/>
      </RetryUntilSuccessful>
      <AlwaysFailure name="Failure 1"
                     _successIf="variable"/>
      <Sequence>
          ...rest of my code...
      </Sequence>
    </ReactiveSequence>

```

If there is a better way to achieve this, I am all ears :)

facontidavide, 2023-06-14:
This is my proposed solution:

- Move the `RetryUntilSuccessful` outside the reactive loop
- We need an instance pf `VariableUpdaterConditionNode` inside the Reactive loop for updating.
- This is correct `<AlwaysFailure name="Failure 2"  _successIf="variable"/>`, but it is shorter to use `_while` in `Sequence`.

```xml
<Sequence>
    <RetryUntilSuccessful name="Retry 1"  num_attempts="-1"   _skipIf="variable ">
        <VariableUpdaterConditionNode topic_name="/test_topic"  vehicle="{variable}"/>
    </RetryUntilSuccessful>

    <ReactiveSequence>
      <VariableUpdaterConditionNode topic_name="/test_topic" vehicle="{variable}"/>
      <Sequence _while="variable">
          ...rest of my code...
      </Sequence>
    </ReactiveSequence>
</Sequence>
```

closed 2023-06-14 by facontidavide