BehaviorTree/BehaviorTree.CPP #587: BT.CPP 4.3.0 RetryUntil

kd-ufr, 2023-06-14:


SteveMacenski, 2023-06-14:
^ adding a description might be useful friend :-)

kd-ufr, 2023-06-14:
fat fingers! I'm editing now sorry

facontidavide, 2023-06-14:
Even if I think that you might have found a bug (thanks!) at the logical level, this tree makes no sense, because you will **ALWAYS** have an infinite loop in "Retry 2".

> In Example A, I expect the tree to tick Retry 1 and Success 1 once, and then skip them in subsequent ticks of the tree. 

In that case, there is no one stopping this code from executing forever:

``` xml
<RetryUntilSuccessful name="Retry 2"    num_attempts="-1">
          <AlwaysFailure name="Failure 2"/>
</RetryUntilSuccessful>
``` 

Anyway, I agree that, even if the logic itself seems wrong, "Retry 1" should be executed once. I will fix it.

kd-ufr, 2023-06-14:
Thanks for getting back to me. And yes, Retry 2 is an infinite loop. In my working code I have a big subtree that returns RUNNING, but I was trying to make a small runable example to demonstrate the behaviour I was seeing. Thanks again.

facontidavide, 2023-06-14:
Push fixed. thanks for reporting

closed 2023-06-14 by facontidavide