BehaviorTree/BehaviorTree.CPP #588: Potential memory overflow in simple_string.hpp

OPerepadia, 2023-06-15:
Recently I started seeing these warnings during compilation, however there are no any errors. I'm not sure, but seems there is some memory overflow in simple_string.hpp.

```bash
In file included from /home/oleksandr/git/BehaviorTree.CPP/include/behaviortree_cpp/utils/convert_impl.hpp:17,
                 from /home/oleksandr/git/BehaviorTree.CPP/include/behaviortree_cpp/utils/safe_any.hpp:27,
                 from /home/oleksandr/git/BehaviorTree.CPP/include/behaviortree_cpp/scripting/any_types.hpp:28,
                 from /home/oleksandr/git/BehaviorTree.CPP/include/behaviortree_cpp/scripting/operators.hpp:23,
                 from /home/oleksandr/git/BehaviorTree.CPP/tests/script_parser_test.cpp:3:
In member function ‘void SafeAny::SimpleString::createImpl(const char*, std::size_t)’,
    inlined from ‘void SafeAny::SimpleString::createImpl(const char*, std::size_t)’ at /home/oleksandr/git/BehaviorTree.CPP/include/behaviortree_cpp/utils/simple_string.hpp:147:8,
    inlined from ‘SafeAny::SimpleString::SimpleString(const char*, std::size_t)’ at /home/oleksandr/git/BehaviorTree.CPP/include/behaviortree_cpp/utils/simple_string.hpp:54:15,
    inlined from ‘SafeAny::SimpleString::SimpleString(const SafeAny::SimpleString&)’ at /home/oleksandr/git/BehaviorTree.CPP/include/behaviortree_cpp/utils/simple_string.hpp:26:83,
    inlined from ‘T BT::Any::cast() const [with T = SafeAny::SimpleString]’ at /home/oleksandr/git/BehaviorTree.CPP/include/behaviortree_cpp/utils/safe_any.hpp:213:34:
/home/oleksandr/git/BehaviorTree.CPP/include/behaviortree_cpp/utils/simple_string.hpp:153:18: warning: ‘void* memcpy(void*, const void*, size_t)’ writing between 18446744073709551504 and 18446744073709551615 bytes into a region of size 0 [-Wstringop-overflow=]
  153 |       std::memcpy(_storage.str.data, input_data, size);
      |       ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/oleksandr/git/BehaviorTree.CPP/include/behaviortree_cpp/utils/simple_string.hpp:152:44: note: destination object of size 0 allocated by ‘operator new []’
  152 |       _storage.str.data = new char[size + 1];
      |                                            ^
/home/oleksandr/git/BehaviorTree.CPP/include/behaviortree_cpp/utils/simple_string.hpp:154:31: warning: writing 1 byte into a region of size 0 [-Wstringop-overflow=]
  154 |       _storage.str.data[size] = '\0';
      |       ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/home/oleksandr/git/BehaviorTree.CPP/include/behaviortree_cpp/utils/simple_string.hpp:152:44: note: at offset [-112, -1] into destination object of size 0 allocated by ‘operator new []’
  152 |       _storage.str.data = new char[size + 1];
      |                                            ^
/home/oleksandr/git/BehaviorTree.CPP/include/behaviortree_cpp/utils/simple_string.hpp:152:44: warning: argument 1 range [18446744073709551505, 18446744073709551615] exceeds maximum object size 9223372036854775807 [-Walloc-size-larger-than=]
In file included from /usr/include/c++/13.1.1/bits/new_allocator.h:34,
                 from /usr/include/c++/13.1.1/x86_64-pc-linux-gnu/bits/c++allocator.h:33,
                 from /usr/include/c++/13.1.1/bits/allocator.h:46,
                 from /usr/include/c++/13.1.1/memory:65,
                 from /usr/local/include/gtest/gtest.h:57,
                 from /home/oleksandr/git/BehaviorTree.CPP/tests/script_parser_test.cpp:1:
/usr/include/c++/13.1.1/new: In member function ‘T BT::Any::cast() const [with T = SafeAny::SimpleString]’:
/usr/include/c++/13.1.1/new:128:26: note: in a call to allocation function ‘void* operator new [](std::size_t)’ declared here
  128 | _GLIBCXX_NODISCARD void* operator new[](std::size_t) _GLIBCXX_THROW (std::bad_alloc)
      |                          ^~~~~~~~
[100%] Linking CXX executable behaviortree_cpp_test
[100%] Built target behaviortree_cpp_test
```

facontidavide, 2023-06-18:
The error is telling you that the **size** (second argument) is somehow wrong.

This is not a problem of the class, but your input instead, I think.

If you believe it is an issue of the class, please send a way to reproduce the issue.

OPerepadia, 2023-06-18:
> Do you see this error during the compilation of the library? On which platform?

Yes, I see it during compilation of the library.

OS: Arch Linux
platform: x86_64

If you want to reproduce the issue, I created this Dockerfile:

```Dockerfile
FROM archlinux

WORKDIR /usr/src/app

RUN pacman -Syu --noconfirm && \
    pacman -S base-devel git cmake zeromq sqlite gtest --noconfirm
    
RUN git clone https://github.com/BehaviorTree/BehaviorTree.CPP.git && \
    cd BehaviorTree.CPP && \
    mkdir build && cd build && cmake .. && make -j$(nproc)
```

You can build it with:
```bash
docker build -t bt_cpp .
```

aboseley, 2023-10-03:
I'm also seeing this using fedora:38, using the above steps (using gcc 13.2.1).  The warnings only occurs when compiling with optimizations (-DCMAKE_BUILD_TYPE=Release), which is a concern.  This usually points to some non-compliant code that gcc is optimizing.