BehaviorTree/BehaviorTree.CPP #589: Update README

Lecrapouille, 2023-06-16:
The compilation steps in the README do not seem correct.

```
mkdir build; cd build
conan install ../BehaviorTree.CPP --output-folder=. --build=missing
cmake ../BehaviorTree.CPP -DCMAKE_TOOLCHAIN_FILE=conan_toolchain.cmake
cmake --build . --parallel
```

The correct steps are:
```
mkdir build; cd build
conan install .. --output-folder=. --build=missing
cmake .. -DCMAKE_TOOLCHAIN_FILE=conan_toolchain.cmake
cmake --build . --parallel
```

SteveMacenski, 2023-06-16:
You can submit that as a PR you know :smile:

Lecrapouille, 2023-06-16:
@SteveMacenski For such simple changes PRs are overkill but I made one :)

SteveMacenski, 2023-06-16:
I disagree, but not providing the PR, you're making a simple change that Davide could just hit "merge" on into something he has to himself make the changes onto and PR himself. You've already identified the patch, by submitting the PR you're saving everyone time in the long run

facontidavide, 2023-06-18:
I think that:
- You missed the line "Assuming that you are in the **parent** directory of `BehaviorTree.CPP`"
- I used "../" instead of "./"

closed 2023-06-18 by facontidavide

Lecrapouille, 2023-06-18:
Correct ! I was more accustomed to have the build folder inside the project