BehaviorTree/BehaviorTree.CPP #591: Tutorial not compiling (v3.8)

Lecrapouille, 2023-06-17:
Hi!

The initial tutorial "Your first Behavior Tree" for version 3.8 (HEAD)
https://www.behaviortree.dev/docs/3.8/tutorial-basics/tutorial_01_first_tree

seems not compiling:
```
g++ --std=c++17 -Wall -Wextra BT.cpp -o prog -lbehaviortree_cpp_v3
BT.cpp: In function ‘int main()’:
BT.cpp:65:82: error: cannot convert ‘main()::<lambda()>’ to ‘const TickFunctor&’ {aka ‘const std::function<BT::NodeStatus(BT::TreeNode&)>&’}
   65 |   factory.registerSimpleCondition("CheckBattery", [&](){ return CheckBattery(); });
      |                                                                                  ^
In file included from BT.cpp:1:
/usr/local/include/behaviortree_cpp_v3/bt_factory.h:293:72: note:   initializing argument 2 of ‘void BT::BehaviorTreeFactory::registerSimpleCondition(const string&, const TickFunctor&, BT::PortsList)’
  293 |                                const SimpleConditionNode::TickFunctor& tick_functor,
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
BT.cpp:69:79: error: cannot convert ‘main()::<lambda()>’ to ‘const TickFunctor&’ {aka ‘const std::function<BT::NodeStatus(BT::TreeNode&)>&’}
   69 |   factory.registerSimpleAction("OpenGripper", [&](){ return gripper.open(); } );
      |                                                                               ^
In file included from BT.cpp:1:
/usr/local/include/behaviortree_cpp_v3/bt_factory.h:282:66: note:   initializing argument 2 of ‘void BT::BehaviorTreeFactory::registerSimpleAction(const string&, const TickFunctor&, BT::PortsList)’
  282 |                             const SimpleActionNode::TickFunctor& tick_functor,
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
BT.cpp:70:80: error: expected ‘)’ before ‘auto’
   70 |   factory.registerSimpleAction("CloseGripper", [&](){ return gripper.close(); }
      |                               ~                                                ^
      |                                                                                )
......
   77 |    auto tree = factory.createTreeFromFile("./my_tree.xml");
      |    ~~~~                                                                         
BT.cpp:83:3: error: ‘tree’ was not declared in this scope; did you mean ‘free’?
   83 |   tree.tickRootWhileRunning();
      |   ^~~~
      |   free
```

It seems we have to update this tutorial statement:

```
The only requirement of the functor is to have either one of these signatures:
BT::NodeStatus myFunction()
BT::NodeStatus myFunction(BT::TreeNode& self) 
```

Only the second function is allowed.
(and `ApproachObject` compiles well).

Currently, I do not know how to pass this information. In addition in the main() function the tree is created after theses definitions. We have probably to "swap" lines.

I will check how to fix that and create a PR but can you confirm first ?

Lecrapouille, 2023-06-17:
Ok I did not noticed first the cpp code of tutorial we have to replace `[&](){` by `std::bind` I will create a pull request on the other repo

facontidavide, 2023-06-18:
fixed differently

closed 2023-06-18 by facontidavide