BehaviorTree/BehaviorTree.CPP #593: BT.CPP 4.3.1 Skipping a parallel node's child

kd-ufr, 2023-06-22:
Hi Davide,

Following up from #587, I have found that children of parallel nodes are not skipped consistently. I've included a simple example below:

```
  <BehaviorTree ID="TestTree">
    <Sequence>
      <Script code="test := true"/>
      <Parallel failure_count="1"
                success_count="-1">
        <KeepRunningUntilFailure name="Run 1"
                                 _skipIf="test == true">
          <AlwaysFailure name="Fail 1"/>
        </KeepRunningUntilFailure>
        <KeepRunningUntilFailure name="Run 2">
          <AlwaysSuccess name="Success 1"/>
        </KeepRunningUntilFailure>
      </Parallel>
    </Sequence>
  </BehaviorTree>
```

I'd expect 'Run 1' to always be skipped. However, it is only skipped on the first tick, and executes on subsequent ticks:
```
Sequence                  IDLE -> RUNNING
Script                    IDLE -> SUCCESS
Parallel                  IDLE -> RUNNING
Run 1                     IDLE -> SKIPPED
Run 2                     IDLE -> RUNNING
Success 2                 IDLE -> SUCCESS
Success 2                 SUCCESS -> IDLE
Run 1                     SKIPPED -> RUNNING
Fail 1                    IDLE -> FAILURE
Fail 1                    FAILURE -> IDLE
Run 1                     RUNNING -> FAILURE
Run 1                     FAILURE -> IDLE
Run 2                     RUNNING -> IDLE
Parallel                  RUNNING -> FAILURE
Script                    SUCCESS -> IDLE
Parallel                  FAILURE -> IDLE
Sequence                  RUNNING -> FAILURE
Sequence                  FAILURE -> IDLE
```

It seems like 'Run 1' isn't being moved to IDLE so it's precondition isn't assessed again. Please advise if this is intended behaviour or not.

Thanks for your help!

facontidavide, 2023-06-22:
is this version 3.8?

facontidavide, 2023-06-22:
fixed!

closed 2023-06-22 by facontidavide

kd-ufr, 2023-06-22:
Thanks Davide :)