BehaviorTree/BehaviorTree.CPP #595: Type error with  _skipif

PolarisXQ, 2023-06-25:
The aim of the simple tree is to control the robot according to the message published on a ROS2 topic. I want to read messages from ROS2 topics, write them on the blackboard, and then decide whether some action should be executed or not based on the variables' values.

This is how the Topics2Blackboard Node is defined (clip that may be related):
```
#Topics2Blackboard.cpp
PortsList Topics2Blackboard::providedPorts()
  {
    return {
      OutputPort<bool>("base_front_remain"),
      OutputPort<std::uint16_t>("my_sentry_hp"),
      OutputPort<std::uint16_t>("my_base_hp"),
      OutputPort<std::uint16_t>("enemy_sentry_hp"),
      OutputPort<std::uint16_t>("enemy_base_hp"),
      OutputPort<std::uint8_t>("game_progress"),
      OutputPort<std::uint16_t>("stage_remain_time"),
      OutputPort<std::uint8_t>("armor_id"),
      OutputPort<std::uint8_t>("hurt_type"),
      OutputPort<std::uint8_t>("find_enemy"),
      // OutputPort<nav_msgs::msg::Odometry>("self_odom"),
      OutputPort<nav_msgs::msg::Odometry>("target_odom"),
    };
  }

......

void Topics2Blackboard::target_callback_(const auto_aim_interfaces::msg::Target::SharedPtr msg)
  {
    target_ = *msg;
    setOutput("find_enemy", 0);//A value for test
    if(target_.tracking){
      try {
            tf_buffer_->transform<geometry_msgs::msg::Point>(
              target_.position,target_r_map_, to_frame_, tf2::durationFromSec(0.1));
          } catch (const tf2::TransformException & ex) {
            RCLCPP_INFO(
              node_->get_logger(), "Could not transform %s to %s: %s",
              to_frame_.c_str(), target_.header.frame_id.c_str(), ex.what());
            return;
          }
          target_odom_.header.stamp = target_.header.stamp;
          target_odom_.header.frame_id = to_frame_;
          target_odom_.pose.pose.position.x = target_r_map_.x;
          target_odom_.pose.pose.position.y = target_r_map_.y;
          target_odom_.pose.pose.position.z = target_r_map_.z;
          setOutput<nav_msgs::msg::Odometry>("target_odom", target_odom_);
    }
  }
```

and this is my tree.xml
```
<?xml version="1.0" encoding="UTF-8"?>
<root BTCPP_format="4">
  <BehaviorTree ID="MainTree">
    <ReactiveSequence>
      <Topics2Blackboard armor_id="{armor_id}"
                         base_front_remain="{base_front_remain}"
                         enemy_base_hp="{enemy_base_hp}"
                         enemy_sentry_hp="{enemy_sentry_hp}"
                         find_enemy="{find_enemy}"
                         game_progress="{game_progress}"
                         hurt_type="{hurt_type}"
                         my_base_hp="{my_base_hp}"
                         my_sentry_hp="{my_sentry_hp}"
                         stage_remain_time="{stage_remain_time}"
                         target_odom="{target_odom}"/>
      <ReactiveSequence>
        <CalculatePose goal="{goal}"
                       _skipIf="find_enemy==0"/>
        <Nav2Pose goal="{goal}"
                  _skipIf="find_enemy==0"/>
      </ReactiveSequence>
    </ReactiveSequence>
  </BehaviorTree>

  <!-- Description of Node Models (used by Groot) -->
  <TreeNodesModel>
    <Action ID="CalculatePose"
            editable="true">
      <output_port name="goal"/>
    </Action>
    <Action ID="Nav2Pose"
            editable="true">
      <input_port name="goal"/>
    </Action>
    <Action ID="Topics2Blackboard"
            editable="true">
      <output_port name="armor_id"/>
      <output_port name="base_front_remain"/>
      <output_port name="enemy_base_hp"/>
      <output_port name="enemy_sentry_hp"/>
      <output_port name="find_enemy"/>
      <output_port name="game_progress"/>
      <output_port name="hurt_type"/>
      <output_port name="my_base_hp"/>
      <output_port name="my_sentry_hp"/>
      <output_port name="stage_remain_time"/>
      <output_port name="target_odom"/>
    </Action>
  </TreeNodesModel>

</root>
```

But I got this error
```
[tree_exec_node-32] terminate called after throwing an instance of 'std::runtime_error'
[tree_exec_node-32]   what():  Can't mix different types in ExprComparison
```

facontidavide, 2023-06-25:
I am investigating.

But this doesn't look a great implementation:

```xml
      <ReactiveSequence>
        <CalculatePose goal="{goal}"
                       _skipIf="find_enemy==0"/>
        <Nav2Pose goal="{goal}"
                  _skipIf="find_enemy==0"/>
      </ReactiveSequence>
```

I would rather do 

```xml
      <ReactiveSequence  _skipIf="find_enemy==0">
        <CalculatePose goal="{goal}"   />
        <Nav2Pose goal="{goal}" />
      </ReactiveSequence>
```
Also, do you want to recalculate pose at each single tick? is this the reason why you use **ReactiveSequence**?

facontidavide, 2023-06-25:
I created a branch called **issue_595**

Would you be so kind to test it?

PolarisXQ, 2023-06-26:
The tree now looks like this:
```
  <BehaviorTree ID="test">
    <ReactiveSequence>
      <Topics2Blackboard armor_id="{armor_id}"
                         base_front_remain="{base_front_remain}"
                         enemy_base_hp="{enemy_base_hp}"
                         enemy_sentry_hp="{enemy_sentry_hp}"
                         find_enemy="{find_enemy}"
                         game_progress="{game_progress}"
                         hurt_type="{hurt_type}"
                         my_base_hp="{my_base_hp}"
                         my_sentry_hp="{my_sentry_hp}"
                         stage_remain_time="{stage_remain_time}"
                         target_odom="{target_odom}"/>
      <Sequence _skipIf="find_enemy==0">
        <CalculatePose goal="{goal}"/>
        <Nav2Pose goal="{goal}"/>
      </Sequence>
    </ReactiveSequence>
  </BehaviorTree>
```
I got this when I checkout to issue_595 and rebuilt this package
```bash
[tree_exec_node-32] terminate called after throwing an instance of 'std::runtime_error'
[tree_exec_node-32]   what():  Can't mix different types in ExprComparison. Left operand [unsigned char] right operand [long]
```
so I tried to add more types into the isNumber function in safe_any.hpp, but it didn't work.

PolarisXQ, 2023-06-26:
I changed the type of OutputPort into long for test
```cpp
  PortsList Topics2Blackboard::providedPorts()
  {
    return {
      OutputPort<bool>("base_front_remain"),
      OutputPort<std::uint16_t>("my_sentry_hp"),
      OutputPort<std::uint16_t>("my_base_hp"),
      OutputPort<std::uint16_t>("enemy_sentry_hp"),
      OutputPort<std::uint16_t>("enemy_base_hp"),
      OutputPort<std::uint8_t>("game_progress"),
      OutputPort<std::uint16_t>("stage_remain_time"),
      OutputPort<std::uint8_t>("armor_id"),
      OutputPort<std::uint8_t>("hurt_type"),
      OutputPort<long>("find_enemy"),//change here
      // OutputPort<nav_msgs::msg::Odometry>("self_odom"),
      OutputPort<nav_msgs::msg::Odometry>("target_odom"),

    };
  }

.......

void Topics2Blackboard::target_callback_(const auto_aim_interfaces::msg::Target::SharedPtr msg)
  {
    target_ = *msg;
    setOutput<long>("find_enemy", long(0)); //change here
    if(target_.tracking){
      try {
            tf_buffer_->transform<geometry_msgs::msg::Point>(
              target_.position,target_r_map_, to_frame_, tf2::durationFromSec(0.1));
          } catch (const tf2::TransformException & ex) {
            RCLCPP_INFO(
              node_->get_logger(), "Could not transform %s to %s: %s",
              to_frame_.c_str(), target_.header.frame_id.c_str(), ex.what());
            return;
          }
          target_odom_.header.stamp = target_.header.stamp;
          target_odom_.header.frame_id = to_frame_;
          target_odom_.pose.pose.position.x = target_r_map_.x;
          target_odom_.pose.pose.position.y = target_r_map_.y;
          target_odom_.pose.pose.position.z = target_r_map_.z;
          setOutput<nav_msgs::msg::Odometry>("target_odom", target_odom_);
    }
  }
```

and this time:
```bash
[tree_exec_node-32] terminate called after throwing an instance of 'std::runtime_error'
[tree_exec_node-32]   what():  Any::cast failed because it is empty
```

facontidavide, 2023-06-26:
Thanks for the feedback, I will continue working on this until it solved! I will let you know if I make any progress

facontidavide, 2023-06-26:
For the records, I have no idea why my test passes and you code works

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/issue_595/tests/script_parser_test.cpp#L335-L358

facontidavide, 2023-06-26:
This makes me think that non initialized variables are too fragile.

If `find_enemy` has never been initialized, the **skipIf** will always throw an exception that the user can not catch.

A solution would be to consider a non-initialized blackboard entry similar to **not-a-number,** i.e. all the comparison operators will always fail.

What do you think?

facontidavide, 2023-06-26:
I improved the error messages. With the latest code, the exceptions should be much more informative (pull latest code)

PolarisXQ, 2023-06-27:
I moved the code ```setOutput<long>("find_enemy", long(0));``` into tick(), and it worked! So probably there was something wrong with the error message. 
With the latest code, the error reporting of uninitialized variables is accurate!
Thank you so much!!

PolarisXQ, 2023-06-27:
It would be better to recommend the user use c++17 std::optional to cope with this problem. This was mentioned in document V3.8, but not in the latest one.

commit 236bfee8 (2023-06-27): fix #595 : improvement in blackboard/scripting types (#597)
closed 2023-06-27 by PolarisXQ