BehaviorTree/BehaviorTree.CPP #601: BT.CPP 4.3.2 onHalted not run when halting parallel children

kd-ufr, 2023-06-29:
Hi Davide,

I am finding that when a parallel node halts a running child, its post conditions are not applied the way I expected. I have included an example below:

This tree starts and then halts a RunForever action node. The PrintString then prints the value of variable `test`.
```
  <BehaviorTree ID="test_tree">
    <Sequence>
      <Script code="test := 'start'"/>
      <Fallback>
        <Parallel failure_count="1"
                  success_count="-1">
          <RunForever _onHalted="test = 'halted'"
                                    _post="test = 'post'"/>
          <AlwaysFailure/>
        </Parallel>
        <PrintString key="{test}"/>
      </Fallback>
    </Sequence>
  </BehaviorTree>
```

I would have expected the `onHalted` condition to run and 'halted' to be printed. However, 'post' was printed, indicating the `post` condition was run. If I remove the `post` condition, 'start' is printed, indicating `onHalted` is not being run at all. 

Thanks for your help!

facontidavide, 2023-07-05:
Problem fixed. thanks for reporting.

[image]

closed 2023-07-05 by facontidavide

kd-ufr, 2023-07-05:
Awesome, thanks Davide

commit 43075b50 (2023-07-05): bug fix #601: onHalted not called correctly in Control Nodes