BehaviorTree/BehaviorTree.CPP #605: Passing blackboard value to a subtree can change its type

alsora, 2023-07-12:
I observed a very strange (and hard to debug) type-conversion issue around subtree ports and parsed scripts.

I'm running with BehaviorTree.cpp version 4.0.1

The following code can reproduce the problem
```
<BehaviorTree ID="MySubtree">
      <Script code=" my_value:=false " />
</BehaviorTree>

<BehaviorTree ID="MyMainTree">
    <Control ID="Sequence">
            <Script code=" my_value:=true " />
            <SubTree ID="MySubtree" my_value="{my_value}  "/>
    </Control ID="Sequence">
</BehaviorTree>
```

This is the error, which happens when executing the script inside the subtree, with a gdb backtrace

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Any::copyInto fails
--Type <RET> for more, q to quit, c to continue without paging--

Thread 1 "my-task" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffefd4ca700 (LWP 265618)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) 
(gdb) 
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff0be8859 in __GI_abort () at abort.c:79
#2  0x00007ffff0fc2911 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff0fce38c in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff0fce3f7 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff0fce6a9 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x0000555557b6686c in BT::Any::copyInto(BT::Any&) ()
#7  0x0000555557b68b2b in BT::Ast::ExprAssignment::evaluate(BT::Ast::Environment&) const ()
#8  0x0000555557b52e1d in  ()
#9  0x0000555557b39eab in BT::ScriptNode::tick() ()
#10 0x0000555557b517aa in BT::TreeNode::executeTick() ()
#11 0x0000555557b24640 in BT::SyncActionNode::executeTick() ()
#12 0x0000555557b517aa in BT::TreeNode::executeTick() ()
#13 0x0000555557b4c363 in BT::DecoratorNode::executeTick() ()
#14 0x00007ffeff032c07 in Sequence::tick() ()
#15 0x0000555557b517aa in BT::TreeNode::executeTick() ()
#16 0x0000555557b2c2a3 in BT::Tree::tickRoot(BT::Tree::TickOption, std::chrono::duration<long, std::ratio<1l, 1000l> >) ()
```

It took me a good amount of time to find why the code was throwing and come up with a minimal reproducible example.
The subtree is defined as follows
```
 <SubTree ID="MySubtree" my_value="{my_value}  "/>
```

due to a typo, there's a whitespace between the `}` and the `"` and this seems to cause the type to change

facontidavide, 2023-07-12:
Hi, for my own sanity, I will investigate only issues that are reproducible in the master branch.

Can you tell me if 4.3.3 is affected?

facontidavide, 2023-07-13:
Fixed.
Your problem is that you added whitespaces in the remapping

```xml
<SubTree ID="MySubtree" my_value="{my_value}  "/>
```

When it should be 
```xml
<SubTree ID="MySubtree" my_value="{my_value}"/>
```

This means that it is recognized as a string and remapping fails.
I fixed that and also improved the error message to be less cryptic.

closed 2023-07-13 by facontidavide