BehaviorTree/BehaviorTree.CPP #619: `AsyncSequence` in 4.3.4 does not return execution

alsora, 2023-07-27:
Hi,

I just updated my BehaviorTree.cpp library from 4.0.1 to 4.3.4.

Reading the changelog I saw that the `Sequence` behavior has been reverted to a previous state, and a new `AsyncSequence` has been introduced.

Given the following BT
```
                <root BTCPP_format="4">
                <BehaviorTree ID="MainTree">
                    <Control ID="AsyncSequence">
                        <Decorator ID="Repeat" num_cycles="5">
                            <Action ID="AlwaysSuccess"/>
                        </Decorator>
                    </Control>
                </BehaviorTree>
                </root>
```

In 4.0.1 using a `Sequence` node I have that I need to tick the tree multiple times in order to complete the execution (i.e. the `Repeat` node returns `RUNNING` 5 times).
In 4.3.4 I would expect the same behavior using the `AsyncSequence`, however that's not the case.

The tree completes in a single tick.

Is this expected?

Thank you

facontidavide, 2023-08-01:
This is expected (but not necessarily desired).

**AsyncSequence** does not (and can not) change the behavior of **Repeat**.

Therefore, what you will need is an **AsyncRepeat** here, or modify **Repeat** to make it interruptable.

alsora, 2023-08-01:
Thank you for the answer.
I agree that `AsyncSequence` shouldn't affect his children.
However, it looks like "something" changed in how nodes are processed and I'm trying to understand if (and how) I need to modify my behavior trees in order to safetly update BehaviorTree.cpp to the latest version.

As previously mentioned, running that BT I have two different behaviors depending on the version: in 4.0.1 the tree tick returns after every iteration of the `Repeat` (let's call this async-behavior), while in 4.3.4 it does not (let's call this sync-behavior).

I did some more tests.
I backported the 4.3.4 `AsyncSequence` to 4.0.1 and it works as expected (i.e. I get the async-behavior).
If I use the 4.0.1 `Sequence` in 4.3.4, this presents the problematic, sync-behavior.

I added some more logs using the 4.3.4 `AsyncSequence` node in BT.cpp 4.0.1 and BT.cpp 4.3.4
The tree is always the one posted above.
See here a few lines from the logs (truncated)

Running in 4.0.1
```
-----------> Application is ticking the tree
AsyncSequence: tick() is called
[1690904050.338]: AsyncSequence              IDLE -> RUNNING
AsyncSequence: calling tick on child node
[1690904050.338]: AlwaysSuccess             IDLE -> SUCCESS
[1690904050.338]: Repeat                    IDLE -> RUNNING
-----------> Application is ticking the tree
AsyncSequence: tick() is called
AsyncSequence: calling tick on child node
[1690904050.338]: AsyncSequence             IDLE -> SUCCESS
-----------> Application is ticking the tree
AsyncSequence: tick() is called
AsyncSequence: calling tick on child node
[1690904050.338]: AlwaysSuccess             IDLE -> SUCCESS
....
```

Running in 4.3.4
```
-----------> Application is ticking the tree
AsyncSequence: tick() is called
[1690904655.077]: AsyncSequence              IDLE -> RUNNING
AsyncSequence: calling tick on child node
[1690904655.077]: Repeat                    IDLE -> RUNNING
[1690904655.077]: AlwaysSuccess             IDLE -> SUCCESS
[1690904655.077]: AlwaysSuccess             SUCCESS -> IDLE
AsyncSequence: tick() is called
AsyncSequence: calling tick on child node
[1690904655.077]: AlwaysSuccess             IDLE -> SUCCESS
[1690904655.077]: AlwaysSuccess             SUCCESS -> IDLE
AsyncSequence: tick() is called
AsyncSequence: calling tick on child node
[1690904655.077]: AlwaysSuccess             IDLE -> SUCCESS
[1690904655.077]: AlwaysSuccess             SUCCESS -> IDLE
....
```

Could you help me understand what is causing the different behavior?
Thanks

facontidavide, 2023-08-01:
Yes, probably something DID change in the node Repeat. 

Most probably, I reverted its asynchronous behavior introduced in 4.0.0.

If I understand you correctly, you expect this code:

```xml
<root BTCPP_format="4">
  <BehaviorTree ID="MainTree">
    <Repeat num_cycles="5">
      <AlwaysSuccess/>
    </Repeat>
   </BehaviorTree>
</root>
``` 
To be asynchronous and require 5 ticks. Is that correct?

I think I agree with this BUT, I can see some users complaining that they use **Repeat** in their **ReactiveSequence** and this breaks their code (this happened already).

A potential solution would be to create a decorator **ForceSynchronous** that people using **ReactriveSequence** may adopt for ther "condition" branch.

What do you think?

facontidavide, 2023-08-01:
**UPDATE:**

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/src/decorators/repeat_node.cpp#L68C1-L74C10

No, repeat is supposed to be asynchronous!

It must be a bug in the code!

I am not sure if I have much time to investigate it this week, but I will try to fix it the next week.

alsora, 2023-08-02:
Thank you for the information.
Indeed I could reproduce the problem just by using the `Repeat` node, without any `Sequence` or `AsyncSequence`.

Upon further investigation, I think that the difference in behavior between the two versions is caused by this PR https://github.com/BehaviorTree/BehaviorTree.CPP/pull/522

In my code I'm using `tree->tickOnce()` which now has been replaced by `tree->tickExactlyOnce()`.

At this point I think that this isn't really a bug, but rather an expected behavior?

facontidavide, 2023-08-08:
yes, you are right. That is an expected behavior.
But thanks for taking the time to investigate this!

closed 2023-08-08 by facontidavide