BehaviorTree/BehaviorTree.CPP #622: PublisherZMQ is never built

MFHava, 2023-08-01:
Are you aware that ever since [acdc6e9](https://github.com/BehaviorTree/BehaviorTree.CPP/commit/acdc6e998ab3dc0d878da18793ee48e03282ce30#diff-1e7de1ae2d059d21e1dd75d5812d5a34b0222cef273b7c3a2af62eb747f9d20aL64-L72) your build system makes no attempt at compiling PublisherZMQ (`bt_zmq_publisher.cpp`) whilst said class is still provided in your API leading to linker errors for anyone including the respective header?

facontidavide, 2023-08-01:
This class is not supported anymore in 4.x, only in 3.x

It should be removed entirety

MFHava, 2023-08-01:
You may consider it unsupported, but the fact remains that it is in the repo, is part of the public API and had me spent like half a day trying to figure out the ensuing linker error because I wouldn't expect a source file from the repo not being compiled...

facontidavide, 2023-08-01:
I apologize for that. I will remove it in 4.3.5

closed 2023-08-01 by facontidavide