BehaviorTree/BehaviorTree.CPP #623: SubTreePlus in v3.8.4: Port initially created with type [std::string] and, later type [...] was used somewhere else

m-morelli, 2023-08-01:
Ciao @facontidavide,

I've recently upgraded BehaviorTree.CPP from v3.8.3 to v3.8.4. Now when my BT executor executes a `SubTreePlus` like
```
<SubTreePlus name="Patrol" ID="Patrol" tl1="KITCHEN" tl2="LIVING"/>
```
I get this error:
```
[ERROR] [1690826385.151509220] []: Original error: The creation of the tree failed because the port [tl1] was initially created with type [std::string] and, later type [std_msgs::msg::String_<std::allocator<void> >] was used somewhere else.
```

_It was used to work in v3.8.3_. Possibly related to https://github.com/BehaviorTree/BehaviorTree.CPP/issues/489, which you already fixed for BehaviorTree.CPP v4?

________

Here's a simple test that reproduces the issue, based on the `MoveBase` node in BehaviorTree.CPP for your convenience:
```C++
#include "behaviortree_cpp_v3/loggers/bt_cout_logger.h"
#include "behaviortree_cpp_v3/bt_factory.h"

#include "movebase_node.h"

// clang-format off

static const char* xml_text = R"(
<root main_tree_to_execute="Test">
  <BehaviorTree ID="Test">
    <ReactiveSequence name="MainSequence">
      <SubTreePlus name="Visit2" ID="Visit2" tl1="1;2;3" tl2="4;5;6"/>
    </ReactiveSequence>
  </BehaviorTree>
  <BehaviorTree ID="Visit2">
    <Sequence name="Visit2MainSequence">
      <Action name="MoveBase" ID="MoveBase" goal="{tl1}"/>
      <Action name="MoveBase" ID="MoveBase" goal="{tl2}"/>
    </Sequence>
  </BehaviorTree>
</root>
 )";

// clang-format on

using namespace BT;

int main()
{
  BehaviorTreeFactory factory;
  factory.registerNodeType<MoveBaseAction>("MoveBase");
  auto tree = factory.createTreeFromText(xml_text);
  tree.tickRootWhileRunning();
  return 0;
}
```

Execution on my system gives:
```
$ ./test_subtreeplus 
tl2 (std::string) -> full
tl1 (std::string) -> full
terminate called after throwing an instance of 'BT::RuntimeError'
  what():  The creation of the tree failed because the port [tl1] was initially created with type [std::string] and, later type [Pose2D] was used somewhere else.
Aborted (core dumped)
```

siferati, 2023-08-04:
Having the same issue after updating from 3.8.3 to 3.8.4 today

facontidavide, 2023-08-14:
fixed

closed 2023-08-14 by facontidavide