BehaviorTree/BehaviorTree.CPP #636: SequenceWithMemory skips nodes

daBaret, 2023-08-16:
Hi,
I'm running in to a strange behavior where I'm not sure if my implementation is wrong or if there is a bug.
As you can see I have the following tree:
[image]

The CommandRobots node succeeds and the CheckRobotsState will fail until it reaches the desired state. When it succeeds the setBlackboard node is called. For some reason when setBlackBoard is called it skips all the next one without executing them. So the LogNode is not called

I tried with the Sequence instead of SequenceWithMemory and in that case it works.

Do you know why this happens?
Thank you
Davide

facontidavide, 2023-08-28:
Please create a small example / unit test that I can run that reproduce the problem and I will be happy to address this as soon as I can

facontidavide, 2023-08-31:
Just added a unit test, but I can not reproduce your issue.

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/tests/gtest_sequence.cpp#L393-L431

Closing this, until you provide a reproducible example

daBaret, 2023-08-31:
Ehi, I'm sorry that I didn't answer anymore.
I noticed that it fails only with `SetBlackboard` node.
Test:
```bash
TEST(SequenceWithMemoryTest, Issue_636)
{
  static const char* xml_text = R"(

<root BTCPP_format="4" main_tree_to_execute="MainTree" >

    <BehaviorTree ID="MainTree">
        <SequenceWithMemory>
            <Script code = " var := 0 " />
            <ScriptCondition code = "var+=1; var >= 5" />
            <SetBlackboard value="42" output_key="value" />
            <TestB/>
            <TestC/>
        </SequenceWithMemory>
    </BehaviorTree>
</root> )";

  BT::BehaviorTreeFactory factory;

  std::array<int, 3> counters;
  RegisterTestTick(factory, "Test", counters);

  auto tree = factory.createTreeFromText(xml_text);

  auto res = tree.tickOnce();
  int tick_count = 1;

  while(res != BT::NodeStatus::SUCCESS)
  {
    res = tree.tickOnce();
    tick_count++;
  }

  ASSERT_EQ(1, counters[0]);
  ASSERT_EQ(1, counters[1]);
  ASSERT_EQ(1, counters[2]);

  ASSERT_EQ(5, tick_count);
}
```
Could also be that the test I created is wrong. Could you confirm?
Thank you for your time

closed 2023-09-05 by facontidavide