BehaviorTree/BehaviorTree.CPP #637: Connection with Groot2: connect once then lose connection

ipa-rwu, 2023-08-17:
I am running BT in loops and want to monitoring via Groot2. 
First I try as an example [here]
https://github.com/BehaviorTree/BehaviorTree.CPP/blob/78bdb97b122d685d0b3b8c89bffa402f97998748/examples/generate_log.cpp#L31
Then I found it connected once, then lose connection.

I think it need to call "flush()" function
https://github.com/BehaviorTree/BehaviorTree.CPP/blob/78bdb97b122d685d0b3b8c89bffa402f97998748/include/behaviortree_cpp/loggers/groot2_publisher.h#L55
But it is defined under private.

facontidavide, 2023-08-17:
As you can see here, this is not your problem:

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/src/loggers/groot2_publisher.cpp#L228-L231

> Then I found it connected once, then lose connection.

It must be something else. Can you provide more details? I am assuming that you are using Groot2.
Could it be a connection problem?

facontidavide, 2024-05-08:
close because no more details where provided

closed 2024-05-08 by facontidavide